ngageoint/six-library #21: Building on OS

piyushrpt, 2015-10-08:
Hi,
  I had no trouble building the package on an Ubuntu VM but am having trouble building the library on OS-X (macports - gcc48). 

To get started, I had to modify   *projects/csm/wscript*
```
osxRegex = r'darwin'
```
section to allow it to configure correctly. 

However, I run into the following error when building the package:
```
[ 94/722] cxx: externals/coda-oss/modules/c++/tiff/source/Header.cpp -> target/externals/coda-oss/modules/c++/tiff/source/Header.cpp.1.o
../externals/coda-oss/modules/c++/sys/source/OSUnix.cpp: In member function 'virtual void sys::OSUnix::getMemInfo(size_t&, size_t&) const':
../externals/coda-oss/modules/c++/sys/source/OSUnix.cpp:279:45: error: '_SC_PHYS_PAGES' was not declared in this scope
     long long totalNumPages = sysconfCaller(_SC_PHYS_PAGES);
                                             ^
../externals/coda-oss/modules/c++/sys/source/OSUnix.cpp:280:45: error: '_SC_AVPHYS_PAGES' was not declared in this scope
     long long availNumPages = sysconfCaller(_SC_AVPHYS_PAGES);
                                             ^
../externals/coda-oss/modules/c++/tiff/source/Common.cpp: In function 'void tiff::split(sys::Uint64_T, sys::Uint32_T&, sys::Uint32_T&)':
../externals/coda-oss/modules/c++/tiff/source/Common.cpp:57:44: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     numerator = ((sys::Uint32_T *)&value)[0];
```

Any suggestions?

Thanks
Piyush

chvink, 2015-10-08:
Yeah, those two symbols don't exist on OSX from what I can tell. Unfortunately I don't have access to a machine running OSX at the moment so getting a fix working may be kind of painful. Take a look at the `osx_compat` branch I just pushed, this should hopefully add code for OSX to use. If this successfully compiles, can you do `./waf install --targets=OSTest`, then run the program `${PREFIX}/tests/sys/OSTest`? That should call the function that was causing problems, can you make sure it's printing reasonable values for system memory?

Thanks

piyushrpt, 2015-10-08:
Came across this  for *_SC_PHYS_PAGES_* when trying to solve the issue before posting here.

http://stackoverflow.com/questions/30511579/linux-sc-phys-pages-not-working-on-mac-os-x

I couldn't find anything similar for *_SC_AVPHYS_PAGES_* .

Here is a diff of the wscript:
```
@@ -5,6 +5,7 @@ PLUGIN             = 'CSM'
 REMOVEPLUGINPREFIX = True

 import sys, os, re
+from waflib import Errors

 DIRS = 'external'

@@ -22,6 +23,7 @@ def configure(conf):
     linuxRegex = r'.*-.*-linux-.*|i686-pc-.*|linux'
     solarisRegex = r'sparc-sun.*|i.86-pc-solaris.*|sunos'
     winRegex = r'win32'
+    osxRegex = r'darwin'
     if re.match(linuxRegex, sys.platform):
         platformName = 'linux'
     elif re.match(solarisRegex, sys.platform):
@@ -37,6 +39,8 @@ def configure(conf):
         #       For example, if you're building with VS 2008 Express, use:
         #       --enable-32bit "--msvc_version=msvc 9.0Exp"
         platformName = 'winVC' + str(conf.env['MSVC_VERSION']).replace('.', '')
+    elif re.match(osxRegex, sys.platform):
+        platformName='darwin'
     else:
         raise Errors.WafError('Unsupported platform %s' % sys.platform)
```

Will check out the new branch and report back.

Thanks

piyushrpt, 2015-10-08:
I had to make a minor change to get this to work
```
--- a/externals/coda-oss/modules/c++/sys/source/OSUnix.cpp
+++ b/externals/coda-oss/modules/c++/sys/source/OSUnix.cpp
@@ -291,7 +291,7 @@ void sys::OSUnix::getMemInfo(size_t &totalPhysMem, size_t &freePhysMem) const
 #if defined(__APPLE__)
     long long physMem = 0;
     size_t size = sizeof(physMem);
-    int status = sysctlbyname("hw.memsize", &physMem, &length, 0, 0);
+    int status = sysctlbyname("hw.memsize", &physMem, &size, 0, 0);
     if(status)
     {
         throw sys::SystemException(Ctxt("Call to sysctl() has failed"));
```

The code compiled fine after that and reported reasonable values:

```
Platform: Darwin (x86_64): 12.6.0 [build: Darwin Kernel Version 12.6.0: Wed Mar 18 16:23:48 PDT 2015; root:xnu-2050.48.19~1/RELEASE_X86_64]
Num CPUs: 8
The delimiter on this platform: /
The process id: 55951
Total physical memory (MB): 16384
Total free memory (MB): 6181
Did not find file: /Users/dummy/packages/six-library/blah.txt
Found directory: /Users/dummy/packages/six-library
Created file: ./TMPOMXeqM with size: 35
Killed file: ./TMPOMXeqM
Created test directory: TEST_DIRECTORY
Time Now: 1444329235
Start: 1.44433e+12
Finish Loop 1
Finish Loop 2
Finish Loop 3
Stop: 0.448974609375
Time Now: 1444329235
```

Piyush

chvink, 2015-10-08:
Thanks for your help! I'll put in pull requests for these changes.

Charles

closed 2015-10-08 by chvink