ngageoint/six-library #23: Caster for size_t missing/ broken

piyushrpt, 2015-10-15:
Looks like the casting from nitf::Field to size_t appears to be missing / broken.

```
[596/722] cxx: modules/c++/six/source/ReadControlFactory.cpp -> target/modules/c++/six/source/ReadControlFactory.cpp.1.o
../modules/c++/six/source/NITFReadControl.cpp: In member function 'void six::NITFReadControl::load(nitf::IOInterface&, const std::vector<std::basic_string<char> >&)':
../modules/c++/six/source/NITFReadControl.cpp:152:77: error: conversion from 'nitf::Field' to 'size_t {aka long unsigned int}' is ambiguous
                         subheaderLength(subheader.getSubheaderFieldsLength());
```

chvink, 2015-10-15:
Hmm, I'm not getting a compiler error there, and I'm having trouble reproducing that ambiguity in isolation, too. Are you still on macports gcc 4.8? Can you provide the `waf` command you used to configure and compile?

Thanks

piyushrpt, 2015-10-15:
I'm still on macports and gcc48. Here are the commands:

For configuring (I also tried enable-cpp11):
```
./waf configure --prefix=install --with-cflags="-I /opt/local/include" --with-cxxflags="-I /opt/local/include" --with-linkflags="-L /opt/local/lib"
```

For building:
```
./waf
```

Thanks

piyushrpt, 2015-10-15:
This may have something to do with it: 

http://stackoverflow.com/questions/11603818/why-is-there-ambiguity-between-uint32-t-and-uint64-t-when-using-size-t-on-mac-os

asylvest, 2015-10-15:
Does changing that line to

```
const size_t subheaderLength = subheader.getSubheaderFieldsLength();
```

compile?  If not, how about 

```
const size_t subheaderLength = static_cast<size_t>(subheader.getSubheaderFieldsLength());
```

or

```
const size_t subheaderLength = (size_t)subheader.getSubheaderFieldsLength();
```

Or is it really `size_t` specific such that this works

```
const nitf::Uint64 subheaderLength(subheader.getSubheaderFieldsLength());
```

piyushrpt, 2015-10-15:
I think it really is size_t, nitf::Uint64 (last option) worked but the other ones did not. 
This went past the previous error, but generated another error:

```
[668/722] cxx: projects/csm/external/vts_301_0/source/vts_misc.cpp -> target/projects/csm/external/vts_301_0/source/vts_misc.cpp.2.o
../projects/csm/external/vts_301_0/source/vts_misc.cpp:30:20: fatal error: termio.h: No such file or directory
 #include <termio.h>
                    ^
compilation terminated.
```

This link might be relevant: https://developer.apple.com/library/mac/documentation/Porting/Conceptual/PortingUnix/compiling/compiling.html

Unfortunately, I develop on an OS X laptop. It's easier to test and play with six if I can build it on my laptop.

Thanks

asylvest, 2015-10-16:
Yeah we want SIX to be buildable anywhere so we're happy to support... unfortunately, I don't think I have easy access to an OS X machine myself to test anything so this will be somewhat trial and error.

I pushed up a fix to change that `size_t` (and the underlying constant) to an explicit `sys::Uint64_T`.

vts is actually not something we wrote; it's a 3rd party tool that we bundle with SIX for convenience (it's used to test out the CSM plugins) so we don't want to change the source code there (and I'm not sure who / if anyone maintains this software now to pass this onto them).  The easiest way around this is to configure with the `--disable-csm` flag which will then not try to build anything in the projects/csm directory.  Another way to avoid this would be that you can explicitly call out the applications that you want to build.  More than likely, this would cover everything you care about: `waf install --target=six-samples` (you could also do `waf install --target=six-samples,six.sicd-python` if you care about Python bindings too).  If you do actually want to build the CSM plugin, it would be pretty quick for us to add a configure flag to disable vts from the default `waf install` build (or equivalently you could comment out lines 44-49 of projects/csm/external/wscript).

If you do skip building the vts stuff, let me know if you can make it through the rest of the SIX build successfully or if you run into any other compilation hiccups.

piyushrpt, 2015-10-16:
I tried proceeding without csm. Looks like I ran into a swig / waf issue: 

```
[659/703] cxxshlib: target/externals/coda-oss/modules/python/except/source/generated/coda_except_wrap.cxx.3.o -> target/externals/coda-oss/modules/python/except/_coda_except.so
ld: unknown option: -h_coda_except.so

Build failed
 -> task in 'except-python' failed (exit status 1): 
	{task 4491107088: cxxshlib coda_except_wrap.cxx.3.o -> _coda_except.so}
['/opt/local/bin/g++', '-L', '/opt/local/lib', '-fPIC', '', '-Wl,-headerpad_max_install_names', '-bundle', '-undefined', 'dynamic_lookup', '-Wl,-headerpad_max_install_names', '-bundle', '-undefined', 'dynamic_lookup', '-Wl,-h_coda_except.so', 'externals/coda-oss/modules/python/except/source/generated/coda_except_wrap.cxx.3.o', '-o', '/Users/dummy/packages/six-library/target/externals/coda-oss/modules/python/except/_coda_except.so', '-Lexternals/coda-oss/modules/c++/except', '-lexcept-c++.1.0', '-L/opt/local/lib', '-L/opt/local/lib/db48']
```

Thanks

chvink, 2015-10-16:
Sounds like your version of `ld` is >2.17. Usage of `-h` changes at some point after that version; I'll see what we can do to get around this.

asylvest, 2015-10-16:
In the meantime, you can skip building the Python stuff (if you just care about the C++ bindings) by configuring with `--disable-python`.

piyushrpt, 2015-10-16:
The *--disable-python* option works. But I would be interested in the python bindings since I'm trying to test this for use with the ISCE package.

Thanks

chvink, 2015-10-16:
Take a look at the `python_ld_update` branch I just pushed, that should fix the extra logic involved in compiling the python shared objects.

piyushrpt, 2015-10-16:
Removing the -h did not solve the problem. Looks like this might be relevant - replacing soname with install_name on OS X 

http://stackoverflow.com/questions/4580789/ld-unknown-option-soname-on-os-x

piyushrpt, 2015-10-16:
Had to make following OS X specific changes to make progress:

```
--- a/externals/coda-oss/build/build.py
+++ b/externals/coda-oss/build/build.py
@@ -1304,8 +1304,8 @@ def process_swig_linkage(tsk):
     #       ld version <= 2.17)
     libpattern = tsk.env['cshlib_PATTERN']
     linkarg_pattern = '-Wl,%s'
-    rpath_pattern = '-Wl,-rpath=%s'
-    soname_pattern = '-soname=%s'
+    rpath_pattern = '-Wl,-rpath,%s'
+    soname_pattern = '-install_name,%s'
```

Failed in another OS X specific error:
```
[672/698] cxxshlib: target/externals/coda-oss/modules/python/math.linear/source/generated/math_linear_wrap.cxx.3.o -> target/externals/coda-oss/modules/python/math.linear/_math_linear.so
ld: can't link with bundle (MH_BUNDLE) only dylibs (MH_DYLIB) file 'externals/coda-oss/modules/python/except/_coda_except.so' for architecture x86_64
collect2: error: ld returned 1 exit status
.....
Build failed
 -> task in 'math.linear-python' failed (exit status 1):
	{task 4353742096: cxxshlib math_linear_wrap.cxx.3.o -> _math_linear.so}
['/opt/local/bin/g++', '-L', '/opt/local/lib', '-fPIC', '', '-Wl,-headerpad_max_install_names', '-bundle', '-undefined', 'dynamic_lookup', '-Wl,-headerpad_max_install_names', '-bundle', '-undefined', 'dynamic_lookup', '-pthread', 'externals/coda-oss/modules/python/except/_coda_except.so', '-Wl,-install_name,_math_linear.so', '-Wl,-rpath,:${ORIGIN}/../coda', 'externals/coda-oss/modules/python/math.linear/source/generated/math_linear_wrap.cxx.3.o', '-o', '/Users/dummy/packages/test/six-library/target/externals/coda-oss/modules/python/math.linear/_math_linear.so', '-Lexternals/coda-oss/modules/c++/math.linear', '-Lexternals/coda-oss/modules/c++/sys', '-Lexternals/coda-oss/modules/c++/str', '-Lexternals/coda-oss/modules/c++/except', '-lmath.linear-c++.0.2', '-lsys-c++.1.2', '-lstr-c++.1.0', '-lexcept-c++.1.0', '-Lexternals/coda-oss/modules/python/except', '-L/opt/local/lib', '-L/opt/local/lib/db48', '-ldl']
```

chvink, 2015-10-16:
Can you post the output of `ld -v` or its equivalent? 

It's looking like there's a mixup between OSX bundles and true dynamic libraries. I tried to force g++ to build things correctly with `-dynamiclib`, hopefully that got it. There are updates on the branch.

piyushrpt, 2015-10-16:
Here is my *ld -v*
```
@(#)PROGRAM:ld  PROJECT:ld64-242.2
configured to support archs: i386 x86_64 x86_64h armv6 armv7 armv7s armv7m armv7k arm64
LTO support using: LLVM version 3.5.2
```

When we add -dynamiclib, we get the following error:
```
[656/698] cxxshlib: target/externals/coda-oss/modules/python/except/source/generated/coda_except_wrap.cxx.3.o -> target/externals/coda-oss/modules/python/except/_coda_except.so
g++: error: -bundle not allowed with -dynamiclib
...
Waf: Leaving directory `/Users/dummy/packages/test/six-library/target'
Build failed
 -> task in 'except-python' failed (exit status 1): 
	{task 4501428048: cxxshlib coda_except_wrap.cxx.3.o -> _coda_except.so}
['/opt/local/bin/g++', '-L', '/opt/local/lib', '-fPIC', '', '-Wl,-headerpad_max_install_names', '-bundle', '-undefined', 'dynamic_lookup', '-Wl,-headerpad_max_install_names', '-bundle', '-undefined', 'dynamic_lookup', '-dynamiclib', '-Wl,-install_name,_coda_except.so', 'externals/coda-oss/modules/python/except/source/generated/coda_except_wrap.cxx.3.o', '-o', '/Users/agram/packages/test/six-library/target/externals/coda-oss/modules/python/except/_coda_except.so', '-Lexternals/coda-oss/modules/c++/except', '-lexcept-c++.1.0', '-L/opt/local/lib', '-L/opt/local/lib/db48']
```

I digged into the exact command line a little more. The problem is the macports python by default is not built with gcc / g++. Macports builds this with clang. However, it has no trouble when numpy / scipy and other libraries are built with gcc and linked to libpython . When replace the above call with clang and remove -dynamiclib, everything works fine. Here is the command that I used:

```
/usr/bin/clang -L /opt/local/lib -fPIC  -Wl,-headerpad_max_install_names -bundle -undefined dynamic_lookup -Wl,-headerpad_max_install_names -bundle -undefined dynamic_lookup -Wl,-install_name,_coda_except.so externals/coda-oss/modules/python/except/source/generated/coda_except_wrap.cxx.3.o -o /Users/agram/packages/test/six-library/target/externals/coda-oss/modules/python/except/_coda_except.so -Lexternals/coda-oss/modules/c++/except -lexcept-c++.1.0 -L/opt/local/lib -L/opt/local/lib/db48
```
The compiler used to compile python can be obtained using standard tools and I see a lot of this being printed out in *config.log* during the configure step. I guess using the correct compiler when building the extensions would solve the problem.

asylvest, 2015-10-16:
Hmm, I'm a little hesitant as this may open up another can of works but I wonder if life would be easier if you built SIX using clang.  When you do a `waf configure -h`, for the `--check-c-compiler` and `--check-cxx-compiler` flags, does it list clang?  If you add `--check-c-compiler=clang --check-cxx-compiler=clang`, does it find them?  I am nearly positive that you'll run into something that you'll need to add/fix in the configureCompilerOptions() function inside of build.py to get this to work, so depends on your motivation level (but if you can get that sorted out, we'd merge it into the master).

piyushrpt, 2015-10-16:
I will run some tests and see how far I can get. I suspect that the only section one needs to change is the swig_linkage parts. Replacing env.derive by a copy and changing the compiler to clang (or whatever it is that is gathered during configure step for python) should do the trick, I suspect.

asylvest, 2015-10-16:
Yeah you're right that you could do that... I was just thinking the cleanest solution - rather than mixing and telling waf to use different compilers for different spots - would be to update build.py (and any other source files where clang is pickier) to just support a pure clang compilation.

chvink, 2015-10-19:
>When replace the above call with clang and remove -dynamiclib, everything works fine.

I think it's `-bundle` that we want to remove here; most of our python interface libraries will need to link to other interface libraries at compile time. `coda_except` is one of the few that doesn't depend on any others, but I'm betting that even on clang we'd run into the `can't link with bundle...` error when we get to something like `math.linear-python`. I pushed another update that removes the `-bundle` option in `swig_linkage`, if you haven't already tested something like this.

piyushrpt, 2015-10-19:
I had to make the following changes to get it to work. I still have to disable-csm but the python bindings seem to compile fine.

```
--- a/externals/coda-oss/build/build.py
+++ b/externals/coda-oss/build/build.py
@@ -1290,7 +1290,7 @@ def process_swig_linkage(tsk):
     # flags

     solarisRegex = r'sparc-sun.*|i.86-pc-solaris.*|sunos'
-    darwinRegex = r'darwin'
+    darwinRegex = r'i.86-apple-.*'

@@ -1319,8 +1319,9 @@ def process_swig_linkage(tsk):

     # overrides for osx
     if re.match(darwinRegex,platform):
-        if '-bundle' in tsk.env.LINKFLAGS:
+        while '-bundle' in tsk.env.LINKFLAGS:
             tsk.env.LINKFLAGS.remove('-bundle')
```

asylvest, 2015-10-19:
Thanks - @chvink updated the python_ld_update branch which hopefully incorporates all of this.  Try to use that and let us know if it works properly.  To fully test the Python bindings, try to run test_create_sicd_xml.py.. you'll have to add install/lib/python2.7/site-packages to your PYTHONPATH and set the SIX_SCHEMA_PATH env variable to install/conf/schema/six.  If things work, it should print 'Round trip succeeded!' and you'll have a couple XML files in your directory.

@chvink on this branch can you also update projects/csm/external/wscript to skip building the vts target on Mac?  It sounds like the CSMAPI target as well as the SICD/SIDD plugins are building ok, so it would be just skipping that vts target.  If you put the whole bld.program() block in an if statement, that target will just never show up to waf at all.  That way Mac will build out of the box.

chvink, 2015-10-19:
Done, hopefully that gets everything. (for reference we're still on the python_ld_update branch)

piyushrpt, 2015-10-19:
Builds without csm. Building with csm produces this:

```
[700/703] cxxshlib: target/projects/csm/source/SICDSensorModel.cpp.1.o target/projects/csm/source/SIDDSensorModel.cpp.1.o target/projects/csm/source/SIXPlugin.cpp.1.o target/projects/csm/source/SIXSensorModel.cpp.1.o -> target/projects/csm/SIX_SAR_MDAIS_111_darwin_CSM301.dylib
Undefined symbols for architecture x86_64:
  "csm::GeometricModel::setParameter(int, csm::GeometricModel::Parameter const&)", referenced from:
      vtable for six::CSM::SICDSensorModel in SICDSensorModel.cpp.1.o
      vtable for six::CSM::SIDDSensorModel in SIDDSensorModel.cpp.1.o
      vtable for six::CSM::SIXSensorModel in SIXSensorModel.cpp.1.o
  "csm::CorrelationModel::CorrelationModel(std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long)", referenced from:
      six::CSM::SIXSensorModel::SIXSensorModel() in SIXSensorModel.cpp.1.o
      six::CSM::SIXSensorModel::SIXSensorModel() in SIXSensorModel.cpp.1.o
  "csm::Plugin::theDataDir", referenced from:
      six::CSM::SIXPlugin::constructModelFromState(std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::list<csm::Warning, std::allocator<csm::Warning> >*) const in SIXPlugin.cpp.1.o
      six::CSM::SIXPlugin::constructModelFromISD(csm::Isd const&, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::list<csm::Warning, std::allocator<csm::Warning> >*) const in SIXPlugin.cpp.1.o
  "csm::Plugin::Plugin()", referenced from:
      six::CSM::SIXPlugin::SIXPlugin() in SIXPlugin.cpp.1.o
      six::CSM::SIXPlugin::SIXPlugin() in SIXPlugin.cpp.1.o
      __GLOBAL__sub_I_SIXPlugin.cpp in SIXPlugin.cpp.1.o
  "csm::Version::Version(int, int, int)", referenced from:
      __GLOBAL__sub_I_SICDSensorModel.cpp in SICDSensorModel.cpp.1.o
      __GLOBAL__sub_I_SIDDSensorModel.cpp in SIDDSensorModel.cpp.1.o
      six::CSM::SIXPlugin::getCsmVersion() const in SIXPlugin.cpp.1.o
  "csm::Version::~Version()", referenced from:
      __GLOBAL__sub_I_SICDSensorModel.cpp in SICDSensorModel.cpp.1.o
      __GLOBAL__sub_I_SIDDSensorModel.cpp in SIDDSensorModel.cpp.1.o
  "csm::RasterGM::~RasterGM()", referenced from:
      six::CSM::SIXSensorModel::~SIXSensorModel() in SICDSensorModel.cpp.1.o
      six::CSM::SIXSensorModel::~SIXSensorModel() in SIDDSensorModel.cpp.1.o
      six::CSM::SIXSensorModel::SIXSensorModel() in SIXSensorModel.cpp.1.o
      six::CSM::SIXSensorModel::SIXSensorModel() in SIXSensorModel.cpp.1.o
      six::CSM::SIXSensorModel::~SIXSensorModel() in SIXSensorModel.cpp.1.o
  "csm::GeometricModel::getParameter(int) const", referenced from:
      vtable for six::CSM::SICDSensorModel in SICDSensorModel.cpp.1.o
      vtable for six::CSM::SIDDSensorModel in SIDDSensorModel.cpp.1.o
      vtable for six::CSM::SIXSensorModel in SIXSensorModel.cpp.1.o
  "csm::GeometricModel::getParameterSetIndices(csm::param::Set) const", referenced from:
      six::CSM::SIXSensorModel::getCrossCovarianceMatrix(csm::GeometricModel const&, csm::param::Set, std::vector<csm::GeometricModel const*, std::allocator<csm::GeometricModel const*> > const&) const in SIXSensorModel.cpp.1.o
  "csm::RasterGM::getFamily() const", referenced from:
      vtable for six::CSM::SICDSensorModel in SICDSensorModel.cpp.1.o
      vtable for six::CSM::SIDDSensorModel in SIDDSensorModel.cpp.1.o
      vtable for six::CSM::SIXSensorModel in SIXSensorModel.cpp.1.o
  "typeinfo for csm::RasterGM", referenced from:
      typeinfo for six::CSM::SIXSensorModel in SIXSensorModel.cpp.1.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
...
['/opt/local/bin/g++', '-L', '/opt/local/lib', '-fPIC', '', '-dynamiclib', '-Wl,-compatibility_version,1', '-Wl,-current_version,1', '-pthread', 'projects/csm/source/SICDSensorModel.cpp.1.o', 'projects/csm/source/SIDDSensorModel.cpp.1.o', 'projects/csm/source/SIXPlugin.cpp.1.o', 'projects/csm/source/SIXSensorModel.cpp.1.o', '-o', '/Users/agram/packages/test/six-library/target/projects/csm/SIX_SAR_MDAIS_111_darwin_CSM301.dylib', '-Lmodules/c++/six.sicd', '-Lmodules/c++/six.sidd', '-Lmodules/c++/six', '-Lmodules/c++/scene', '-Lexternals/coda-oss/modules/c++/xml.lite', '-Lexternals/coda-oss/modules/c++/logging', '-Lexternals/coda-oss/modules/c++/tiff', '-Lexternals/coda-oss/modules/c++/io', '-Lexternals/coda-oss/modules/c++/math.linear', '-Lexternals/nitro/modules/c++/nitf', '-Lexternals/coda-oss/modules/c++/mt', '-Lexternals/coda-oss/modules/c++/math', '-Lexternals/coda-oss/modules/c++/sys', '-Lexternals/coda-oss/modules/c++/str', '-Lexternals/coda-oss/modules/c++/except', '-Lexternals/nitro/modules/c/nitf', '-Lexternals/nitro/modules/c/nrt', '-Lexternals/coda-oss/modules/drivers/xml/xerces/xerces-c-3.1.1', '-lsix.sicd-c++.2.1.3', '-lsix.sidd-c++.2.1.3', '-lsix-c++.2.1.3', '-lscene-c++.1.0', '-lxml.lite-c++.1.2', '-llogging-c++.1.1', '-ltiff-c++.1.0', '-lio-c++.1.0', '-lmath.linear-c++.0.2', '-lnitf-c++.2.7', '-lmt-c++.1.1', '-lmath-c++.0.1', '-lsys-c++.1.2', '-lstr-c++.1.0', '-lexcept-c++.1.0', '-lnitf-c.2.7', '-lnrt-c.2.7', '-lxerces-c', '-ldl']
```

piyushrpt, 2015-10-19:
Also, the python bindings appear to be not working (though it compiled without issues). May not be linking to libpython?

```
python ./modules/python/six.sicd/tests/test_create_sicd_xml.py
Traceback (most recent call last):
  File "./modules/python/six.sicd/tests/test_create_sicd_xml.py", line 28, in <module>
    from pysix.scene import *
  File "/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/scene.py", line 28, in <module>
    _scene = swig_import_helper()
  File "/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/scene.py", line 24, in swig_import_helper
    _mod = imp.load_module('_scene', fp, pathname, description)
ImportError: dlopen(/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/_scene.so, 2): Library not loaded: _math_poly.so
  Referenced from: /Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/_scene.so
  Reason: image not found
```

piyushrpt, 2015-10-20:
Just trying the python imports.

All these work
```
import coda
import coda.coda_except
import coda.io
import coda.coda_sys
```

Any submodule that involves importing another submodule fails:
```
import coda.types
ImportError: dlopen(/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/coda/_coda_types.so, 2): Library not loaded: _coda_sys.so
  Referenced from: /Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/coda/_coda_types.so
  Reason: image not found

import coda.math_linear
dlopen(/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/coda/_math_linear.so, 2): Library not loaded: _coda_except.so
  Referenced from: /Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/coda/_math_linear.so
  Reason: image not found
```

asylvest, 2015-10-20:
If you add install/lib to your LD_LIBRARY_PATH, does this fix the problem?  This is what we were trying to avoid needing to do with the various ld flags on Linux, but if this is at least a workaround on Mac, it would let you use the bindings.

It seems like there's a linking problem going on with the CSM stuff too... like it's not linking in the CSMAPI library.

chvink, 2015-10-20:
Something must be messed up with the rpath and/or install_name, I'll look into this. If it's an install_name problem setting the LD_LIBRARY_PATH may not even help.

chvink, 2015-10-20:
I pushed another update to the python_ld_update branch based off of [this](http://stackoverflow.com/questions/4513799/how-to-set-the-runtime-path-rpath-of-an-executable-with-gcc-under-mac-osx) Stack Overflow question. Hopefully the install_name, which should imply rpath, is set correctly now.

piyushrpt, 2015-10-20:
Setting LD_LIBRARY_PATH doesnt help. Let me try with these new changes and get back to you.

piyushrpt, 2015-10-20:
The python bindings seem to be working now, if the order of imports are correct. I only had to include the site-packages to PYTHONPATH. 

We might just have to move the import of dependencies to the very top in math_linear.py and math_poly.py

For example, this fails:
```
from coda import math_linear
```

However
```
from coda import coda_except
from coda import math_linear
```

works. I notice similar behavior with math_poly . For testing, I played around with the Poly1D objects and things seem to be working well.

chvink, 2015-10-20:
What error messages are you given? `from coda import math_linear` is intended to work without having to import coda_except first (in fact the generated python code in `site-packages/coda/math_linear.py` imports coda.coda_except). We definitely don't want users to have to explicitly import all of module's dependencies in order every time they wanted to use it.

piyushrpt, 2015-10-20:
I see that there is an "import coda.coda_except" statement in math_linear.py . Moving this import statement to the very top of math_linear.py resolves the issue. I had to do the same thing with a few others to try and test the python example. 

```
python ~/packages/test/six-library/modules/python/six.sicd/tests/test_create_sicd_xml.py
Traceback (most recent call last):
  File "/Users/dummy/packages/test/six-library/modules/python/six.sicd/tests/test_create_sicd_xml.py", line 29, in <module>
    from pysix.six_sicd import *
  File "/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/six_sicd.py", line 11, in <module>
    import six_base
  File "/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/six_base.py", line 34, in <module>
    _six_base = swig_import_helper()
  File "/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/six_base.py", line 30, in swig_import_helper
    _mod = imp.load_module('_six_base', fp, pathname, description)
ImportError: dlopen(/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/_six_base.so, 2): Library not loaded: @rpath/_xml_lite.so
  Referenced from: /Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/pysix/_six_base.so
  Reason: image not found
```

I noticed in the configure step that waf found xerces and did not build xml lite. But the python bindings still appears to be looking for xml_lite

asylvest, 2015-10-21:
When you're configuring, are you using the `--enable-swig` flag (to re-generate the C++ and Python files which are under the `generated` subdirectories)?  If so, what version of swig are you using?  And, try not using that flag.  We've been using Swig 3.x to generate these for awhile because Swig 1.3.x generates code that the newest versions of gcc won't compile, but we ran into a weird issue today where Swig 1.3.x was also not doing some import stuff quite right in the Python files it generated (but Swig 3.x was fine).

chvink, 2015-10-21:
The _xml_lite.so problem looks more like I used the wrong install_name - I should've used `@rpath/../coda/_xml_lite.so` rather than `@rpath/_xml_lite.so`. This should only affect inter-package dependencies though, something else must be causing the problem with the ordering of imports.

The _xml_lite problem should be fixed with the latest push.

piyushrpt, 2015-10-21:
I noticed the issue with *--enable-swig* and stopped using it after the first couple of attempts. My swig version was 3.0.7 and I noticed that your version was 3.0.5.  I'm using the swig bindings that are provided in the generated subdirs.

Will give the new version a go.

piyushrpt, 2015-10-21:
This did not fix it. Could the problem be in *six-library/modules/python/six/wscript* ? I see xml-lite included as default.

chvink, 2015-10-21:
Hmm. That wscript doesn't do anything system-specific on in and of itself, I don't immediately know if something could be going wrong causing it to fail on OSX but not linux. Have you verified that `/Users/dummy/packages/test/six-library/install/lib/python2.7/site-packages/coda/_xml_lite.so` does not exist? If so, what happens when you run `./waf install --targets=xml.lite-python`? If not, what does `otool -L _xml_lite.so` output?  What does `otool -L _six_base.so` output?

piyushrpt, 2015-10-21:
I experimented by removing xml_lite from wscript and it worked fine. (otool showed that it needed xml_lite. So I removed it from wscript) I was also able to run test_create_sicd_xml successfully, after I made the following changes to the python files - I only moved all the import statements to the very top. 

```
diff --git a/externals/coda-oss/modules/python/math.linear/source/generated/math_linear.py b/externals/coda-oss/modules/python/math.linear/source/generated/math_linear.py
index 0fb5cb9..b371765 100644
--- a/externals/coda-oss/modules/python/math.linear/source/generated/math_linear.py
+++ b/externals/coda-oss/modules/python/math.linear/source/generated/math_linear.py
@@ -6,7 +6,7 @@
 
 
 
-
+import coda.coda_except
 
 from sys import version_info
 if version_info >= (2, 6, 0):
@@ -90,7 +90,7 @@ except AttributeError:
     _newclass = 0
 
 
-import coda.coda_except
+
 class SwigPyIterator(_object):
     """Proxy of C++ swig::SwigPyIterator class"""
     __swig_setmethods__ = {}
diff --git a/externals/coda-oss/modules/python/math.poly/source/generated/math_poly.py b/externals/coda-oss/modules/python/math.poly/source/generated/math_poly.py
index 0af046c..3f3b918 100644
--- a/externals/coda-oss/modules/python/math.poly/source/generated/math_poly.py
+++ b/externals/coda-oss/modules/python/math.poly/source/generated/math_poly.py
@@ -5,7 +5,8 @@
 # the SWIG interface file instead.
 
 
-
+import coda.math_linear
+import coda.coda_except
 
 
 from sys import version_info
@@ -90,8 +91,7 @@ except AttributeError:
     _newclass = 0
 
 
-import coda.math_linear
-import coda.coda_except
+
 
 def new_doubleArray(nelements):
     """new_doubleArray(size_t nelements) -> double *"""
diff --git a/externals/coda-oss/modules/python/types/source/generated/coda_types.py b/externals/coda-oss/modules/python/types/source/generated/coda_types.py
index 0eef2b6..9e2587c 100644
--- a/externals/coda-oss/modules/python/types/source/generated/coda_types.py
+++ b/externals/coda-oss/modules/python/types/source/generated/coda_types.py
@@ -6,7 +6,7 @@
 
 
 
-
+import coda.coda_sys
 
 from sys import version_info
 if version_info >= (2, 6, 0):
@@ -196,7 +196,7 @@ class SwigPyIterator(_object):
 SwigPyIterator_swigregister = _coda_types.SwigPyIterator_swigregister
 SwigPyIterator_swigregister(SwigPyIterator)
 
-import coda.coda_sys
+
 class RowColDouble(_object):
     """Proxy of C++ types::RowCol<(double)> class"""
[... 147 lines cut ...]
index b5b41af..b97df0b 100644
--- a/modules/python/six/wscript
+++ b/modules/python/six/wscript
@@ -10,6 +10,6 @@ def configure(conf):
 def build(bld):
   bld.swigModule(name = 'six',
                  use = 'six-c++ scene-python math.poly-python ' \
-                       'except-python xml.lite-python logging-python ' \
+                       'except-python logging-python ' \
                        'io-python mem-python types-python',
                  package = 'pysix')
diff --git a/wscript b/wscript
index a701d1e..3fa1a7a 100644
--- a/wscript
+++ b/wscript
@@ -27,7 +27,7 @@ def build(bld):
     if not bld.targets:
         bld.targets = 'cphd-c++,six.sicd-c++,six.sidd-c++,six-samples'
         if 'HAVE_CSM' in bld.env:
-            bld.targets += ',six-csm,vts'
+            bld.targets += ',six-csm'
         if bld.is_defined('HAVE_MEX_H'):
             bld.targets += ',nitf_image,nitf_metadata,xml_metadata'
         if 'PYTHON' in bld.env and bld.env['PYTHON'] and bld.is_defined('HAVE_PYTHON_H'):
```

chvink, 2015-10-21:
That's odd, I would not expect that removing dependencies would cause them to get built. I'll keep looking into both of these issues; the python files are automatically generated and we'd like to avoid maintaining separate OSX versions.

piyushrpt, 2015-10-21:
I agree that different versions of python files are inconvenient. Is there anyway to ensure that the import statements end up at the top in the auto-generated py files?

chvink, 2015-10-21:
I'll have to do more research on that, the only way I know of right now is only marginally more maintainable than modifying the generated files.

asylvest, 2015-10-22:
So we'll add this to our backlog of things to take a look at, but unfortunately we'll probably have to set this aside for the time being until we can get a Mac and dig into this directly (but certainly if you're able to figure out something different to do in the .i files or another suggestion we can make the change).  It's unfortunate that you've got to change the auto-generated py files by hand but glad that you do have a way to get the Python bindings working on a Mac.  Thanks for all the back and forth that helped us fix as much as we did so far.

piyushrpt, 2017-06-10:
Hi,
   Finally found time to try building six on a mac again. Using gcc5 and other libraries via macports on an OS-X 10.11.4 machine. Would love to just package all of this into a portfile.

I only got these to build without errors. Haven't actually tested using the python bindings yet, which is the goal.

My configure line: 
```bash
PYTHON=/opt/local/bin/python3 LIBDIR="/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/config-3.5m" python3 waf configure --prefix=/Users/test/packages/MDA/install --with-cxxflags="-I /opt/local/include" --with-cflags="-I /opt/local/include" --with-linkflags="-L/opt/local/lib"
```
I had to make the following other changes to get it to build.

1.  I was able to build the whole library after making the following changes to  build/build.py:
   - sys_platform returns "darwin"
   - libnsl is not needed for darwin
   - Wl,-E option for ld is not supported on darwin 

```python
--- a/externals/coda-oss/build/build.py
+++ b/externals/coda-oss/build/build.py
@@ -801,6 +801,7 @@ def configureCompilerOptions(self):
     linuxRegex = r'.*-.*-linux-.*|i686-pc-.*|linux'
     solarisRegex = r'sparc-sun.*|i.86-pc-solaris.*|sunos'
     winRegex = r'win32'
+    osxRegex = r'darwin'

     cxxCompiler = self.env['COMPILER_CXX']
     ccCompiler = self.env['COMPILER_CC']
@@ -848,7 +849,8 @@ def configureCompilerOptions(self):
     elif ccCompiler == 'gcc' or ccCompiler == 'icc':
         if not re.match(winRegex, sys_platform):
             self.env.append_value('LIB_DL', 'dl')
-            self.env.append_value('LIB_NSL', 'nsl')
+            if not re.match(osxRegex, sys_platform):
+                self.env.append_value('LIB_NSL', 'nsl')
         self.env.append_value('LIB_MATH', 'm')
         self.env.append_value('LINKFLAGS_THREAD', '-pthread')
         self.check_cc(lib='pthread', mandatory=True)
@@ -897,7 +899,8 @@ def configureCompilerOptions(self):
             #       Is there an equivalent to get the same functionality or
             #       is this an OS limitation?
             linkFlags = '-fPIC'
-            if not re.match(solarisRegex, sys_platform):
+
+            if (not re.match(osxRegex, sys_platform)) and (not re.match(solarisRegex, sys_platform)):
                 linkFlags += ' -Wl,-E'

             self.env.append_value('LINKFLAGS', linkFlags.split())
@@ -1311,6 +1314,7 @@ def process_swig_linkage(tsk):

     solarisRegex = r'sparc-sun.*|i.86-pc-solaris.*|sunos'
     darwinRegex = r'i.86-apple-.*'
+    osxRegex = r'darwin'

     platform = getPlatform(default=Options.platform)
     compiler = tsk.env['COMPILER_CXX']
@@ -1338,7 +1342,7 @@ def process_swig_linkage(tsk):
         rpath_pattern = '-Rpath%s'

     # overrides for osx
-    if re.match(darwinRegex,platform):
+    if re.match(darwinRegex,platform) or re.match(osxRegex,platform):
         while '-bundle' in tsk.env.LINKFLAGS:
             tsk.env.LINKFLAGS.remove('-bundle')
         tsk.env.LINKFLAGS.append('-dynamiclib')
```

2. I had to fix a bunch of casts in "six.sicd/source/SICDWriteControl.cpp" and "six/source/NITFWriteControl.cpp" for size_t. I used nitf::Uint64. Can provide exact line numbers if needed. 

3. Replace "termio.h" with "termios.h" in vts_misc.cpp under vts_301_0

4. Map a bunch of things in vts_isd.cpp using define

```c++
#define ftello64 ftello
#define fseeko64 fseeko
#define fopen64 fopen
#define fsetpos64 fsetpos
#define off64_t off_t
#define fpos64_t fpos_t
```

5. Modify six/projects/csm/wscript. Needed linking to libcsmapi.
```python
@@ -68,7 +68,7 @@ def build(bld):
         modArgs = globals()
         modArgs['NAME'] = name
         modArgs['LIBNAME'] = name
-        if re.match(r'win32', sys.platform):
+        if re.match(r'win32', sys.platform) or re.match(r'darwin', sys.platform):
             modArgs['USE'] = 'CSMAPI'
         else:
             modArgs['INCLUDES'] = ['include', bld.env['INCLUDES_CSM']]
```

Piyush

asylvest, 2017-06-10:
Thanks a lot Piyush... can you provide this as a pull request?  If you don't have any need for the CSM stuff for what you're doing, it would probably be simpler to just update the wscript to not try to build the CSM/VTS targets for Mac (we can help do that part).  The thing is that VTS is a 3rd party package and I'd prefer not to have a hand-modified version checked in.  The build.py and cast updates we should be able to take though.

piyushrpt, 2017-06-10:
Cool. Will do. Wasn't sure if my fixes would affect builds on other platforms. I also noticed that I wasn't able to import pysix.scene into python. Looks like the linking paths are hardcoded. Is this in some rpath setting? 

```bash
> otool -L _scene.cpython-35m-darwin.so
_scene.cpython-35m-darwin.so:
	_scene.cpython-35m-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	_math_poly.cpython-35m-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	_math_linear.cpython-35m-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	_coda_types.cpython-35m-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	_coda_sys.cpython-35m-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	_coda_except.cpython-35m-darwin.so (compatibility version 0.0.0, current version 0.0.0)
```

The reference to the coda stuff should have been preceded by a "../coda" ?
Will try the "disable-csm" option. Will be useful to automatically disable it on macs like you said.  

Piyush

piyushrpt, 2017-06-10:
I have issued a pull request for the other changes. It won't build with the "--disable-csm" option. Configure works fine but fails to build. Here is the output waf error :

```bash
  File "/Users/agram/packages/MDA/six-library/externals/coda-oss/build/.waf3-1.7.14-6b92f3c95d8c1bad60826036bf26d63f/waflib/TaskGen.py", line 109, in post
    v()
  File "/Users/agram/packages/MDA/six-library/externals/coda-oss/build/.waf3-1.7.14-6b92f3c95d8c1bad60826036bf26d63f/waflib/Tools/ccroot.py", line 60, in apply_incpaths
    lst=self.to_incnodes(self.to_list(getattr(self,'includes',[]))+self.env['INCLUDES'])
  File "/Users/agram/packages/MDA/six-library/externals/coda-oss/build/.waf3-1.7.14-6b92f3c95d8c1bad60826036bf26d63f/waflib/Tools/ccroot.py", line 37, in to_incnodes
    if x in seen or not x:
TypeError: unhashable type: 'list'

```

piyushrpt, 2017-06-12:
I still can't get the python bindings to work. I suspect it is related to how rpath gets handles on OS X.

This involves changing the rpath with the installation path
[https://stackoverflow.com/questions/4513799/how-to-set-the-runtime-path-rpath-of-an-executable-with-gcc-under-mac-osx](https://stackoverflow.com/questions/4513799/how-to-set-the-runtime-path-rpath-of-an-executable-with-gcc-under-mac-osx)

Alternate solution appears to change the path after installation
[https://blogs.oracle.com/dipol/dynamic-libraries,-rpath,-and-mac-os](https://blogs.oracle.com/dipol/dynamic-libraries,-rpath,-and-mac-os)

Looks like some minor manipulation of the rpath_pattern in build/build.py for coda should fix this?

piyushrpt, 2017-06-13:
I have updated the pull request with changes to build bindings as well on OS X.

asylvest, 2017-06-13:
Thanks for all the work!  I will take a look at this more deeply when I get a chance (may be a few days) and get it merged into the baseline as soon as possible.

piyushrpt, 2018-10-01:
I closed the related pull request. Im able to build six without problems on OS X. Closing this issue as this thread is obsolete now.

closed 2018-10-01 by piyushrpt

ayoungs, 2018-10-01:
It’s also building fine on Mojave 10.14.
Just completed a build.