ngageoint/six-library #37: Build system tries to install j2k_config.h during "build" step

kjurka, 2016-01-06:
With both current git tip and the 2.1.3 release, I see problems building with a target --prefix that the build user cannot access because it's trying to install the j2k_config.h header file.

./waf configure --prefix=/noaccess
./waf build

 in build
    bld.recurse()
  File "/home/kjurka/git/six-library/externals/coda-oss/build/build.py", line 63, in recurse
    os.walk(self.path.abspath()).next()[1]))
  File "/home/kjurka/git/six-library/externals/coda-oss/build/.waf-1.7.14-6b92f3c95d8c1bad60826036bf26d63f/waflib/Context.py", line 125, in recurse
    user_function(self)
  File "/home/kjurka/git/six-library/externals/nitro/modules/c/j2k/wscript", line 84, in build
    makedirs(installDir)
  File "/usr/local/anaconda-2.3/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/usr/local/anaconda-2.3/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/usr/local/anaconda-2.3/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/noaccess'

This installation should only happen during the "install" stage, not "build"

asylvest, 2016-01-07:
Thanks for pointing this out... I agree that this is incorrect behavior.  Looking at the underlying wscript that's being used, the j2k wscript is doing a hack to install the header file rather than the way we usually do it (which is why it's getting installed at build time).

This should be pretty straightforward to fix (the change will actually happen in the NITRO repo - https://github.com/mdaus/nitro - that SIX pulls in); we expect to have a fix for this in the next week or so... I'll keep you posted.

asylvest, 2016-01-09:
Fixed in NITRO... merged in aa2a20722666f797891d039ba98b19c4de2647ba

closed 2016-01-09 by asylvest