ngageoint/six-library #41: python access to region of wideband data

kjurka, 2016-01-20:
As far as I can tell the only way to access the wideband data from python is to read the entire image into memory.  It would be great to have a way to pull out a small chip without allocating space for the entire image.  Or should we be using Nitro directly to deal with non-metadata?

asylvest, 2016-01-21:
You are correct - long story short, the C++ function that we wrap for the Python layer doesn't let you specify an AOI right now.  This is an easy update and something that it makes sense for SIX to provide rather than you needing to wade through NITRO.  Will keep you posted with an update in the near future.

asylvest, 2016-01-22:
@chvink added this functionality - let us know if you run into any issues

closed 2016-02-26 by asylvest