ngageoint/six-library #43: Create __init__.py files on install

JonathanMeans, 2016-01-21:
In order for the Python modules to work (at least on Windows), `__init__.py` files are needed in `install/lib/site-packages/coda` and `install/lib/site-packages/pysix`. Currently, these are not present after a vanilla `waf install` command.

chvink, 2016-01-21:
I'm kind of surprised but it actually worked for me. I just built it with W7, whatever compiler Visual Studio Express 2013 ships with, and python 2.7.9 from anaconda. Until this is figured out you can probably get away with making the files by hand.

mbojrab, 2016-01-21:
I needed to add this to my PYTHONPATH to get it to work, but also was surprised no __init__.py was required. I think its worth adding the __init__.py because its standard in module creation. We should also consider creating a setup.py installer (directly into the python library) or creating a mechanism for installing the egg from the geohub page using pip or something similar. Having a "pip install/update six-library" or "conda isntall/update six-library" would be very beneficial to us and our users. I'd use it.

chvink, 2016-01-21:
I'm getting an `__init__.py` when I built and installed - it's definitely intended to be there. 

I did look into methods of packaging and creating a setup.py a long time ago, but I haven't had the time to follow through.

mbojrab, 2016-01-21:
I just checked my recent six install and the `__init__.py` are populated as expected. <install>\Lib\site-packages\pysix and <install>\Lib\site-packages\coda. It has been some time since I last checked.

I perform one call for my build sequence:
`waf configure --prefix ..\six64.release -o target64.release --disable-warnings install`

JonathanMeans, 2016-01-21:
The __init__.py files are populating for me now, as well. If I did anything differently, I am not aware of it.

The information contained in this communication is confidential, is intended only for the use of the recipient(s) named above, and may be legally privileged. If the reader of this message is not the intended recipient, you are hereby notified that any dissemination, distribution, or copying of this communication is strictly prohibited. If you have received this communication in error, please re-send this communication to the sender and delete the original message or any copy of it from your computer system.

chvink, 2016-02-08:
@JonathanMeans

I just managed to reproduce this problem. I had an issue in a swig binding that caused the build to fail. After fixing that issue, subsequent `./waf install`s would not create the `__init__.py`. A distclean and reconfigure seems to put everything back in working order, but something is not working quite as intended so I'm reopening this issue.

closed 2017-01-23 by asylvest