ngageoint/six-library #50: CPHD writer does not allow classification or release information

clydestanfield, 2016-02-24:
The CPHD writer internally creates a FileHeader object and uses that when writing to disk. This means a user does not have an opportunity to fill out the classification and release information. An override should be created that allows the user to pass in a filled out FileHeader and use that instead.

asylvest, 2016-02-25:
I think the cleanest way to do this is to add two optional string inputs to the writeMetadata() overloading that takes in the pathname and ripple down.  If the string is non-empty, set the FileHeader with that value.  Also while you're in there, fix the `vmb` typos in CPHDWriter.cpp (should be `vbm`).

asylvest, 2016-02-26:
Fixed in #51

closed 2016-02-26 by asylvest