ngageoint/six-library #55: Polarizations of "UNKNOWN" not supported

bombaci-valkyrie, 2016-03-30:
Parsing SICD XML containing polarizations of 'UNKNOWN' cause an "Unsupported conversion" exception.  I first ran into this at [DualPolarizationType six::toType<DualPolarizationType>()](https://github.com/ngageoint/six-library/blob/3f0a57bcfe4e5c3bda77a4cf3a4d5fef679d234f/six/modules/c%2B%2B/six/source/Utilities.cpp#L530), but skimming through the code it looks like 'UNKNOWN' is missing from most of the various polarization routines.

asylvest, 2016-04-04:
Thanks for pointing this out.  FYI, this is mostly fixed under #58... should have it merged to master tomorrow.

asylvest, 2016-04-05:
Resolved under #58 and merged to master - let us know if this doesn't resolve your issue.

closed 2016-04-05 by asylvest