ngageoint/six-library #61: Provide redistributable sample data

bradh, 2016-04-19:
Ideally there would be sample data that allowed wider understanding of SICD. Possibly it could be synthesised from redistributable data in the AFRL data sets (https://www.sdms.afrl.af.mil/).

asylvest, 2016-05-07:
@bradh we'll start gradually putting some samples from public data at https://github.com/ngageoint/six-library/wiki/Sample-SICDs.  For now we've updated with one SICD XML that was generated by converting a public TSX dataset.  If you are interested in the full SICD NITF, not just the XML, we will have to figure out a spot to host the files.

bradh, 2016-05-08:
Thanks for looking at this, and adding the XML. I was hoping for the NITF (in the longer term), but this is still a big head-start.

asylvest, 2016-05-17:
@bradh I've got a spot to host the files now... take a look at the wiki again.  It now contains links to the XML and actual SICD NITF.  We'll add additional examples here down the road as we get a chance.

bradh, 2016-05-18:
Looks good, and is much appreciated.

I'm happy for this issue to be closed, or kept open, whatever works for you.

closed 2016-05-18 by asylvest