ngageoint/six-library #117: Update SIX CSM plugin to support CSM 3.0.3

asylvest, 2016-10-13:


johnpdpkarp, 2016-12-01:
Why would you need to update the SIX CSM plugin for CSM3.0.3 ?  CSM3.0.3 only affects new sensor model families....  The SICD and SIDD CSMs derive from RasterGM which has not changed since CSM3.0.1   ( please see https://github.com/johnpdpkarp for more information on CSM).

asylvest, 2016-12-02:
Ahh that's good news - thanks for letting us know.  We'd heard there was a new CSM version released and wanted to make sure SIX was compatible with it.

I see the CSM code at the link you have above; is there a set of release notes / API changes available somewhere on this page (I didn't see anything in the wiki)?

johnpdpkarp, 2016-12-02:
Currently we have CSM3.0.1 documented on the site.   The naming of this version was somewhat "political".... It really should have just been named 3.0.   Our hope is that NO CHANGES should be occurring to the API defined by 3.0.1, and that anything else we do will simply EXTEND the API.   We don't currently have 3.0.2 on GitHub, but there is a plan to put it there.  3.0.2 introduced a new "family" to support point clouds.  Again, this required no changed to anything coded against 3.0.1  .   3.0.3 also adds a new family for supporting a sensor model for frame sequences (instead of a unique model instance per frame).  The details of that particular "family" are still being worked out and hopefully will be resolved in the next few weeks.  So, the only reason for SIX to be concerned with anything defined beyond 3.0.1 is if you wanted to add a new "family" for supporting volumetric SAR, or something like that... but for anything "raster" based, 3.0.1 is all you need.

closed 2016-12-18 by asylvest