ngageoint/six-library #134: CSM implementation is potentially not handling pixel coordinates correctly

johnpdpkarp, 2016-12-01:
the CSM API is defined such that image coordinates have their integral location at the upper left corner of the image pixel.  I believe the underlying SIX library code uses the center of the pixel as the integral pixel location.    This implies that when projecting coordinates to the ground, the SICD and SIDD CSM implementations need to offset the input pixel locations by (-0.5,-0.5).

asylvest, 2016-12-02:
Thanks for letting us know.  We'll look into this regardless, but was there a specific image or test you did that caused you to be concerned this was incorrect?

@JonathanMeans

johnpdpkarp, 2016-12-02:
I did do some testing and verified my assumption is correct.  I unfortunately can't share the data, sorry!  I don't believe there is necessarily anything wrong with the underlying SIX library code in this regard, just that when mapped to the CSM API, nothing is being done to account for the difference in definitions regarding where the actual "integral" pixel is defined.  From the standpoint of the physical sensor model(s), using the "center of pixel" is the most obvious choice... and pretty much every physical sensor model follows that paradigm.  The original authors of CSM chose the "corner of pixel", so the implemented CSM API method calls need to account for that before calling the underlying math...   John Karp

asylvest, 2016-12-18:
Verified this internally as well.  This is fixed in !137.

closed 2016-12-18 by asylvest