ngageoint/six-library #147: Python CPHD lacks CPHDWriter

darkstar007, 2017-01-29:
Hi,

It appears that the python wrapper for CPHD does not have a CPHDWriter class in it - or have I missed something? This makes writing out CPHD files from python impossible?

Thanks,

Matt

asylvest, 2017-01-30:
Matt - that is correct... when we put in the CPHD bindings, at the time we only needed reading support.  Do you have a current need for writing support?  If so, let me know the urgency and timeframe and we'll try to prioritize it appropriately on our end.

darkstar007, 2017-01-30:
Hi,

I was wanting to write a simple simulator to test the processing chain of our new LF SAR system. So it would be really good if it could be done in the next 6 weeks or so.

Thanks,

Matt

asylvest, 2017-01-30:
Thanks for the timeframe Matt.  We'll see what we can do and will keep you posted...

asylvest, 2017-02-10:
@darkstar007 - see @JonathanMeans's updates in #153.  This has been merged into master... let us know if it meets your needs.

darkstar007, 2017-02-10:
OK, cool. I'll pull and check over the weekend.

darkstar007, 2017-02-17:
Hi.

I've spent some time on this in the last week or so, and have to say its been a hugely frustrating experience. This is mainly due to the lowlevel nature of the API, the lack of documentation and the poor error reporting.

I've yet to actually write out a CPHD file. My current issue is:

```
Traceback (most recent call last):
  File "./sim.py", line 255, in <module>
    writer.writeCPHD(sys.argv[1], vdata, coda_types.RowColSizeT(vdata.shape[0], vdata.shape[1]), vbm.toBuffer(0))
  File "/usr/local/lib/python2.7/dist-packages/pysix/cphd.py", line 2047, in write
    self.write(pathname)
  File "/usr/local/lib/python2.7/dist-packages/pysix/cphd.py", line 2009, in write
    return _cphd.CPHDWriter_write(self, *args)
RuntimeError: Attempted use of uninitialized double value
```

Which tells me I'm missing a variable in the metadata class, but no hints as to which one. So I now need to go through the spec and see what I've missed off.

Can I suggest changing the CPHDWriter.writeCPHD function to be something along the lines of:

```python
def write(self, pathname, vph_data, vbm, channel):
   """
    Function to write out video phase history data to a given filename.
        pathname - the filename (a string) to write the data to
        vph_data - a 2D numpy array (numpy.complex64 type) of size [npulse, nsamps]
        vbm - a pysix.cphd.VBM() like object that describes this data
        channel - the channel number this VPH data is associated with

   Nothing is returned by this function.

   The following exceptions are raise:
           ???
   """
    vphPointer, _ = vph_data.__array_interface__['data']
    
    vbm_raw = vbm.toBuffer(channel)
    vbmPointer, _ = vbm_raw.__array_interface__['data']
    
    if data.dtype != numpy.dtype('complex64'):
        raise TypeError('Python CPHDWriter only supports complex float data')

    # Should we now check that the VPH data we have supplied is the same size that
    # we said it was going to be when we created the CPHDWriter? Similarly for
    # vbm - does it match the metadata and vph_data?

    self.addImageImpl(vphPointer,
                      coda_types.RowColSizeT(vph_data.shape[0], vph_data.shape[1]),
                      vbmPointer)
    self.write(pathname)
```

Obviously I don't know if that's correct, as I've not managed to write a file out yet!
(Also, why does the API talk about addImage - the data isn't an image).

Thanks,

Matt

JonathanMeans, 2017-02-22:
@darkstar007 With #160, you should better be able to tell which fields the writer is complaining about.

darkstar007, 2017-02-25:
Yep, the error message is much more helpful now. Thanks.

closed 2020-11-03 by JDanielSmith