ngageoint/six-library #152: SEGV when printing non-existing antenna item in python

darkstar007, 2017-02-08:
Hi,

If you load a CPHD file that does not have include an antenna structure (which is defined as optional in the specification) and then try to print an element of it, it crashes with a SEGV. I would expect it to return None.

```python
from pysix.cphd import CPHDReader
import multiprocessing
nthreads = multiprocessing.cpu_count()

reader = CPHDReader('myfile.cphd', nthreads)

mdata = reader.getMetadata()

print mdata.antenna.numRcvAnt
```

Just crashes with a SEGV.

Thanks,

Matt

JonathanMeans, 2017-02-08:
What's going on here is that under the hood, `antenna` (and most (all?) optional fields) gets stored as a smart pointer. So for now, you can work around this with an explicit check for `None.` e.g.

    if mdata.antenna.get() is not None:
        print mdata.antenna.numRcvAnt

@asylvest I think I could do something like [this](http://stackoverflow.com/questions/1183716/python-properties-swig) to make the pointer wrappers more Python friendly. The downside there is that I'd have to manually attach it to each type. There's a small chance of magic-ing it all away with reflection, if I have time to experiment.

asylvest, 2017-02-08:
Yeah the crux of it, as @JonathanMeans is driving at, is that in the underlying C++ API, we treat optional parameters as NULL... in the C++ if you're trying to see if an optional field is present, you check if a pointer is NULL.  We generate the Python layer via Swig, so this carries over directly.

We're currently looking into a couple options for making the Python layer more friendly / Pythonic...

darkstar007, 2017-02-10:
OK, cool. Let me know if you want me to test anything

closed 2020-10-12 by JDanielSmith