ngageoint/six-library #169: Could not find a task generator for the name 'six-python'

JBUserGame, 2017-03-19:
I'm trying to build ngaSix.  I did this successfully a few months ago, but now the install is complaining.  I possibly installed a new version of python in the interim.  I first did this:

      `  ./waf configure --prefix=install
`
That seemed to be successful.  But then the install failed:
```
./waf install --targets=six-python
Waf: Entering directory `/home/userjb/src/ngaSIXLibrary/target'
Waf: Leaving directory `/home/userjb/src/ngaSIXLibrary/target'
Could not find a task generator for the name 'six-python'
```
What am I doing wrong?

asylvest, 2017-03-19:
Can you post the output of the `waf configure` command here?  My guess is that `waf` is either not finding Python at all or doesn't like something about it.  When it can't build the Python bindings, it doesn't add the Python targets in which I think is why it can't find a task generator for it.

JBUserGame, 2017-03-19:
Here are the relevant parts of the output of 'waf configure'

```
Checking for program python              : /home/userjb/srcMisc/Python-2.7.13/INSTALL/bin/python
Checking for python version              : 2.7.13 
Checking for library python2.7 in LIBDIR : not found
Checking for library python2.7 in python_LIBPL : not found
Checking for library python2.7 in $prefix/libs : not found
Checking for library python2.7m in LIBDIR      : not found
Checking for library python2.7m in python_LIBPL : not found
Checking for library python2.7m in $prefix/libs : not found
Checking for library python27 in LIBDIR         : not found
Checking for library python27 in python_LIBPL   : not found
Checking for library python27 in $prefix/libs   : not found
Checking for program /home/userjb/srcMisc/Python-2.7.13/INSTALL/bin/python-config,python2.7-config,python-config-2.7,python2.7m-config :
 /home/userjb/srcMisc/Python-2.7.13/INSTALL/bin/python-config
Checking for header Python.h                                                                                                               : :-(
Asking python-config for pyembed --cflags flags                                                                                            : yes
Asking python-config for pyembed --libs flags                                                                                              : yes
Asking python-config for pyembed --ldflags flags                                                                                           : yes
Getting pyembed flags from python-config                                                                                                   : Could not build a python embedded interpreter
The configuration failed
(complete log in /home/userjb/src/Work/Software/ngaSIXLibrary/target/config.log)
Python lib/headers                                                                                                                         : The configuration failed
(complete log in /home/userjb/src/Work/Software/ngaSIXLibrary/target/config.log) (Is python built with -fPIC?) 
```
I'm using a locally installed version of python so it's quite possible something isn't right.

asylvest, 2017-03-19:
Thanks for the log.

Yeah you can take a look at `/home/userjb/src/Work/Software/ngaSIXLibrary/target/config.log` to get a little more detail (you'll see `waf` actually trying to compile the various things that are failing and see the compiler/linker errors).  I'm betting that you'll see a linker error; if that locally installed Python is one you built from source, Python doesn't build with PIC by default.  You'll want to rebuild your Python and in its configure step use `--with-pic` (I think that's the configure option).

JBUserGame, 2017-03-19:
It looks like I have two separate errors in the config.log.  They happen close together.  The first is:
`err: g++: error: unrecognized command line option '-fp-model'`
The second is a g++ compile command that is defining
```
    -DPYTHONDIR="/home/userjb/src/ngaSIXLibrary/install/lib/python2.7/site-packages
    -DPYTHONARCHDIR="/home/userjb/src/ngaSIXLibrary/install/lib/python2.7/site-packages

```
The error message is
`not found`
It advises:
` 
try again with -L$python_LIBPL (some systems don't install the python library in $prefix/lib)
`
But that isn't a waf command-line option.  How can I tell waf where the python libraries are?

JBUserGame, 2017-03-20:
I recompiled python with -fPIC (setenv CFLAGS "-fPIC") and the ngaSIX install was successful.  Thanks for helping with this.

asylvest, 2017-03-20:
OK great!

closed 2017-03-20 by asylvest