ngageoint/six-library #182: msvc Error

mjswartz3141, 2017-05-12:
I'm trying to build SIX without Microsoft Visual studio and am running into this issue

```
c:\dev\six-library-master>python waf configure
Setting top to                           : c:\dev\six-library-master
Setting out to                           : c:\dev\six-library-master\target
Platform                                 : win32
Checking for 'msvc' (c compiler)         : not found
Checking for 'msvc'                      : Warning: cound not find x64 msvc, looking for others
Checking for 'msvc' (c compiler)         : not found
could not configure a c compiler!
(complete log in c:\dev\six-library-master\target\config.log)
```

I've looked into the python build scripts and the second c compiler associated with windows is gcc, which I have sourced in my path:

```
c:\dev\six-library-master>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Anaconda3;C:\Program Files\Anaconda3\Scripts;C:\Program Files\Anaconda3\Library\bin;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;**C:\cygwin64\bin;C:\cygwin64\lib**;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Users\Matthew\AppData\Local\Microsoft\WindowsApps;;C:\Program Files (x86)\Microsoft VS Code\bin
```

(I'm in the process of downloading MSVS, so that's why they've been added to my path) Any idea why the script seems to be searching for msvc twice instead of moving onto gcc? In compiler_c.py, it has

`c_compiler={'win32':['msvc','gcc'],'cygwin':['gcc'],'darwin':['gcc'],'aix':['xlc','gcc'],'linux':['gcc','icc'],'sunos':['suncc','gcc'],'irix':['gcc','irixcc'],'hpux':['gcc'],'gnu':['gcc'],'java':['gcc','msvc','icc'],'default':['gcc'],}`

So I'm confused why msvc pops up twice in the output. Any help would be greatly appreciated!

asylvest, 2017-05-12:
Ahh - so waf automatically prefers using Visual Studio on Windows (it has a default set of compilers it searches per OS) and VS is the only compiler we have ever used on Windows.  You _might_ be able to add `--check-c-compiler=gcc --check-cxx-compiler=g++` to your configure line and configure may succeed.  I'm concerned that we may have some OS-specific checks at configure time that may still take you through a Visual Studio set of configure flags.

Are you using mingw-w64?  If there's a reason you can't use Visual Studio, we can make a best effort attempt to add in support for using gcc on Windows... it's nothing that we've ever needed up to this point so if it's significant work to add, it may be a bit before we have the bandwidth to support.

By the way, if you do end up using Visual Studio, if you're using Win 7 or newer, make sure your command prompt is run as administrator (the way Visual Studio finds its path on the command line annoyingly requires registry access... if you don't do this, you'll see a similar error message to the one you have above).

mjswartz3141, 2017-05-14:
The compiler check flags didn't work and running cmd as an administrator doesn't seem to work either. I have Visual Studio Community (formerly Express Edition) installed and can compile and run C/C++, but the waf configuration seems to not be able to find msvc still. I added the location of cl.exe to my path for both x86 and x64 (C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX86\x86), but am still getting the same issues. 

On another note, I ran `python waf configure` in cygwin, and it ran what looked like successfully, but crashes at the end saying that cygwin is an unsupported platform. 

```
python waf configure
Setting top to                           : /cygdrive/c/dev/six-library-master
Setting out to                           : /cygdrive/c/dev/six-library-master/target
Platform                                 : cygwin
Checking for 'gcc' (c compiler)          : /usr/bin/gcc
Checking for 'g++' (c++ compiler)        : /usr/bin/g++
Checking for library pthread             : yes
Checking for compiler flags -m64         : yes
Checking for compiler flags -m64         : yes
System size                              : 64-bit
Checking for program matlab              : not found
Checking for program python              : /usr/bin/python
Checking for python version              : 2.7.12
Checking for library python2.7 in LIBPATH_PYEMBED : not found
Checking for library python2.7 in LIBDIR          : not found
Checking for library python2.7 in python_LIBPL    : not found
Checking for library python2.7 in $prefix/libs    : not found
Checking for library python2.7m in LIBPATH_PYEMBED : not found
Checking for library python2.7m in LIBDIR          : not found
Checking for library python2.7m in python_LIBPL    : not found
Checking for library python2.7m in $prefix/libs    : not found
Checking for library python27 in LIBPATH_PYEMBED   : not found
Checking for library python27 in LIBDIR            : not found
Checking for library python27 in python_LIBPL      : not found
Checking for library python27 in $prefix/libs      : not found
Checking for program /usr/bin/python-config,python2.7-config,python-config-2.7,python2.7m-config : not found
Checking for header Python.h                                                                     : :-(
Asking python-config for pyembed --cflags flags                                                  : not found
The configuration failed
(complete log in /cygdrive/c/dev/six-library-master/target/config.log)
Python lib/headers                                                                               : The configuration failed
(complete log in /cygdrive/c/dev/six-library-master/target/config.log) (Is python built with -fPIC?)
Checking for CURL                                                                                : not found
Checking for library numpy                                                                       : not found
Configuring with XML layer                                                                       : xerces
Building local lib                                                                               : xerces
Checking for header inttypes.h                                                                   : yes
Checking for header netinet/in.h                                                                 : yes
Checking for header arpa/inet.h                                                                  : yes
Checking for header netdb.h                                                                      : yes
Checking for header intrin.h                                                                     : yes
Checking for header emmintrin.h                                                                  : yes
Checking for header wchar.h                                                                      : yes
Checking for header stdint.h                                                                     : yes
Checking for header stdio.h                                                                      : yes
Checking for header langinfo.h                                                                   : yes
Checking for header iconv.h                                                                      : not found
Checking for header nl_types.h                                                                   : not found
Checking for header ctype.h                                                                      : yes
Checking for header wctype.h                                                                     : yes
Checking for header cpuid.h                                                                      : yes
Checking for header endian.h                                                                     : yes
Checking for header machine/endian.h                                                             : yes
Checking for header arpa/nameser_compat.h                                                        : yes
Checking for header errno.h                                                                      : yes
Checking for header float.h                                                                      : yes
Checking for header locale.h                                                                     : yes
Checking for header sys/param.h                                                                  : yes
[... cut ...]
Checking for function mmap                                                                       : yes
Checking for header stdbool.h                                                                    : yes
Checking for function clock_gettime                                                              : yes
Checking for header sys/time.h                                                                   : yes
Unsupported platform cygwin
```

So it appears that my computer has the necessary components, but running in a command prompt for some reason only looks for msvc.

asylvest, 2017-05-14:
You are doing a good job of finding compilers that we don't support yet. :o)

I updated the build system to build with gcc on Windows.  You will need to add `--check-c-compiler=gcc --check-cxx-compiler=g++` in order for this to work.  The one (actually super annoying) limitation is that when gcc goes to actually run `ar` to create the Xerces library, the command line is > the 32K limit due to our directory structure in SIX.  You can work around this by doing the build in two steps.

First, check out CODA-OSS and ask it to build Xerces (or you can build Xerces yourself):
```
git clone https://github.com/mdaus/coda-oss
cd coda-oss
python waf configure --prefix=install --check-c-compiler=gcc --check-cxx-compiler=g++
python waf install --target=XML
```

Then check out SIX and point it to this Xerces library:
```
git clone https://github.com/ngageoint/six-library.git
cd six-library
python waf configure --prefix=install --check-c-compiler=gcc --check-cxx-compiler=g++ --with-xml-home=\path\to\coda-oss\install
python waf install
```

We'll figure out a better fix for this down the road; I don't really want to change around the directory structure if I can help it.

We've tested with Visual Studio up to 2015 which all build fine.  2017 must have changed something around... we'll look into that when we get a chance.  In the meantime, you should be able to use VS 2015 or gcc (while the gcc code now compiles, I haven't done any testing with it to confirm everything works 100% as expected so VS would be safer if you're using this in production).

asylvest, 2017-05-14:
What was the issue you ran into trying to use VS 2017?  Were you running from a normal command prompt or the VS command prompt?  We try to help waf out to auto-detect VS even from a normal command prompt but it seems to not be working with VS 2017.  However, from a VS 2017 command prompt (specifically the "x64 Native Tools Command Prompt for VS 2017"), waf did find VS 2017 Community Edition on my 64-bit Win 10 machine and successfully built everything.

If you're having a different issue, let me know what you're running into.

mjswartz3141, 2017-05-15:
I am going to be out of town until Wednesday evening, I'll try what you suggested out and report back later in the week. Thanks for your help

mjswartz3141, 2017-05-19:
I tried to build CODA-OSS, but got the following:

```
c:\dev\coda-oss>python waf configure --prefix=install --check-c-compiler=gcc --check-cxx-compiler=g++
Setting top to                           : c:\dev\coda-oss
Setting out to                           : c:\dev\coda-oss\target
Platform                                 : win32
Checking for 'gcc' (c compiler)          : C:\cygwin64\bin\gcc.exe
Checking for 'g++' (c++ compiler)        : C:\cygwin64\bin\g++.exe
Checking for library pthread             : yes
Checking for 64-bit system               : yes
System size                              : 32-bit
Checking for program python              : C:\Program Files\Anaconda3\python.exe
Checking for python version              : 3.6.0
Checking for library python3.6 in $prefix/libs : not found
Checking for library python3.6m in $prefix/libs : not found
Checking for library python36 in $prefix/libs   : not found
Checking for program C:\Program Files\Anaconda3\python.exe-config,python3.6-config,python-config-3.6,python3.6m-config : not found
Checking for header Python.h                                                                                           : :-(
Asking python-config for pyembed --cflags flags                                                                        : not found
The configuration failed
(complete log in c:\dev\coda-oss\target\config.log)
Python lib/headers                                                                                                     : The configuration failed
(complete log in c:\dev\coda-oss\target\config.log) (Is python built with -fPIC?)
Checking for CURL                                                                                                      : not found
Checking for library fftw3                                                                                             : not found
Building local lib                                                                                                     : fftw
Building double precision fft library                                                                                  : no
Checking for header unistd.h                                                                                           : not found
Checking for header getopt.h                                                                                           : not found
Checking for header malloc.h                                                                                           : not found
Checking for header sys/time.h                                                                                         : not found
Checking for function gettimeofday                                                                                     : not found
Checking for function BSDgettimeofday                                                                                  : not found
Checking for function gethrtime                                                                                        : not found
Checking for type hrtime_t                                                                                             : not found
Checking system type sizes                                                                                             : no
The configuration failed
(complete log in c:\dev\coda-oss\target\config.log)
```

I have Python.h in `C:\ProgramData\Microsoft\AndroidNDK64\android-ndk-r13b\prebuilt\windows-x86_64\include\python2.7` as well as in `C:\Program Files\Anaconda3\include` so I think some things are just looking in the wrong place. 

I will be doing most of my development in Visual Studio for debugging ease, so I'm willing to jump ship and get it to work that way, but I ran `python waf configure` in the x64 Native Tools Command Prompt for VS 2017 as an administrator and got the same errors as before. Can you provide your PATH environment variable so I can see if I have anything missing?

Thanks again.

asylvest, 2017-05-22:
The Python errors are actually "ok" in the sense that this will prevent you from building the Python bindings but if you just need the C++ code it won't matter.  It's finding the gcc from Cygwin; the updates I made got the mingw-w64 gcc working... I would not expect Cygwin to work without more updates.

But to your point, let's focus on getting VS working.  Here's the output of my console:

```
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.0.26430.6
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community>echo %PATH%
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX64\x64;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\;C:\Program Files (x86)\Windows Kits\10\bin\x64;C:\Program Files (x86)\Windows Kits\10\bin\10.0.15063.0\x64;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\\MSBuild\15.0\bin;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\Tools\;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Anaconda\2.4.1;C:\Anaconda\2.4.1\Scripts;C:\Anaconda\2.4.1\Library\bin;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\Git\cmd;C:\Program Files\Amazon\AWSCLI\;C:\Program Files\CMake\bin;C:\Program Files\PuTTY\;C:\Users\Adam\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\Docker Toolbox
```

When I run the configure step, it finds VS 2017 as expected:
```
C:\Users\Adam\Documents\Code\checkout\coda-oss>python waf configure --prefix=install
Setting top to                           : C:\Users\Adam\Documents\Code\checkout\coda-oss
Setting out to                           : C:\Users\Adam\Documents\Code\checkout\coda-oss\target
Platform                                 : win32
Checking for 'msvc' (c compiler)         : C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX64\x64\CL.exe
Checking for 'msvc' (c++ compiler)       : C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX64\x64\CL.exe
Checking for 64-bit system               : yes
System size                              : 64-bit
```

If you're still having problems, from the `target` directory that waf generates, can you attach both `config.log` and `waf-print-msvc.bat`?

mjswartz3141, 2017-05-24:
So I successfully ran `python waf configure` on my work machine. I installed Anaconda 2.7, Visual Studio Community 2017, and everything went just fine. When I ran `python waf`, the build failed however. Here is the output:

```
Build failed
 -> task in 'except-python' failed (exit status 1120):
        {task 62208544L: cxxshlib coda_except_wrap.cxx.2.o -> _coda_except.pyd,_coda_except.lib,_coda_except.pyd.manifest}
['C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.10.25017\\bin\\HostX86\\x86\\LINK.exe', '/NOLOGO', '/MANIFEST', '/STACK:80000000', '/MACHINE:X86', '/INCREMENTAL:NO', '/IMPLIB:externals\\coda-oss\\modules\\python\\except\\_coda_except.lib', '/DLL', 'externals\\coda-oss\\modules\\python\\except\\source\\generated\\coda_except_wrap.cxx.2.o', '/OUT:', 'c:\\Users\\mswartz\\dev\\six-library-master\\target\\externals\\coda-oss\\modules\\python\\except\\_coda_except.pyd', '/LIBPATH:externals\\coda-oss\\modules\\c++\\except', 'except-c++.lib', '/LIBPATH:C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.10.25017\\ATLMFC\\lib\\x86', '/LIBPATH:C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.10.25017\\lib\\x86', '/LIBPATH:C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.15063.0\\', '/LIBPATH:C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.15063.0\\', '/LIBPATH:C:\\Windows\\Microsoft.NET\\Framework\\v4.0.30319', '/LIBPATH:C:\\ProgramData\\Anaconda2\\libs', 'python27.lib']
```

Any idea what I did wrong? I didn't include the `--prefix=install` and ran directly from the six-library-master folder.

asylvest, 2017-05-24:
Can you paste more of the surrounding text so I can see what the error is (or redirect all of it to a log file and attach it)?

What aspect of SIX are you trying to use?  If you don't care about the Python bindings, worst case you could just do for example `python waf install --target=round_trip_six` and this would build the C++ dependencies along with this one specific application.  Note that if you don't specify the prefix, I'm not sure offhand where that'll get installed to on Windows but it's possible it's somewhere you don't have permissions to.

The Python bindings do build on Windows on a variety of machines and with various VS versions though, so we can track this down if it's something you want to pursue.

mjswartz3141, 2017-05-24:
See attached. I don't have any interest in the python aspect of SIX, I'm only looking to use the C++. I'll try with the `--prefix` flag and see if that changes anything.
[wafLog.txt](https://github.com/ngageoint/six-library/files/1026460/wafLog.txt)

asylvest, 2017-05-24:
Thanks for the log - I see what's happening.  It looks like you have a 32-bit version of Python installed and are compiling the code with a 64-bit compiler.  That's why you're getting lines like:

```
python27.lib(python27.dll) : fatal error LNK1112: module machine type 'x64' conflicts with target machine type 'X86'

   Creating library externals\coda-oss\modules\python\mem\_mem.lib and object externals\coda-oss\modules\python\mem\_mem.exp
```

So, to get around that you would want to either install a 64-bit version of Python and use that on the configure step or build your code 32-bit (you can do this at configure time with `--enable-32bit`).

But in your case, since you don't actually care about Python at all, at configure time you can add the `--disable-python` flag and waf will simply not try to build any Python code at all.  Then a simple `waf install` should build all of the C++ libraries and applications.

mjswartz3141, 2017-05-24:
I got further with reconfiguring using `--prefix=install`, but the build still failed. See the attached log.
[wafLog.txt](https://github.com/ngageoint/six-library/files/1026565/wafLog.txt)

mjswartz3141, 2017-05-24:
Adam,

I'll give that a try.

Matt

asylvest, 2017-05-24:
Yeah the build failed for the same reason as the earlier log (errors during trying to link in the Python libs).  Let me know how it goes with `--disable-python`.

mjswartz3141, 2017-05-24:
Aha! Running `python waf configure --prefix=install --disable-python` worked and `python waf` built successfully! Onto the next steps! Thanks for the help, I will report back with any new problems.

asylvest, 2017-05-24:
Great!

closed 2017-05-24 by asylvest