ngageoint/six-library #189: Visual Studio Solution

mjswartz3141, 2017-05-25:
Now that I've gotten SIX to build and install, I'm at a loss in getting any of the test programs to run. Is there a developers guide for getting up and running in Visual Studio? Would it be possible to get a MSVS 2008 and older compatible .sln to start off of? Or at least a guide to start a new project (or existing project like I tried and failed from `C:\...\six-library-master\six\modules\c++\samples`) that can run these test programs?

asylvest, 2017-05-25:
After `waf install` finishes, all the binaries are already built and installed in the `install/bin` or `install/tests` folder... you don't actually need to build anything else in order to run the test programs.

If you're interested in writing your own application that links in the SIX library, you can create a VS solution and just add the various SIX libraries in like any other 3rd party dependency. 
 See https://github.com/ngageoint/six-library/raw/master/docs/six-manual.pdf for some additional details, but in short...

It's recommended you use `python waf dumpenv --target=six.sicd-c++,six.sidd-c++` and add these defines to your project.

To get all the libraries that you depend on and the right link order, use `python waf dumplib --target=six.sicd-c++,six.sidd-c++`.  These are what you want to add to your project.

After you do `waf configure`, you can run `waf msvs` which will actually generate a VS solution for you.  This has been tested up to VS 2015 (haven't tried with 2017).  But... this doesn't work quite as nicely as you'd want it to.  It makes a separate VS project for each `waf` target which in this case is about ~600 projects.  It is functional but we don't use it a ton so you may run into issues here... the general recommendation is to do the approach I described earlier.

GitGeoCoder, 2019-03-18:
@asylvest I am trying to build CSM 3.0.3 SICD/SIDD sensor model using VS2017/64-bits. 
Followed the  instructions in https://github.com/ngageoint/six-library/raw/master/docs/six-manual.pdf 

First, the manual is outdated. There is no waf file in ../six/modules, should start the build in the base directory instead ../six

After run 
.\git\six-library-master>python waf configure --prefix=install --enable-64bit
My configuration is 
Setting top to                           : D:\TrustedData\git\six-library-master
Setting out to                           : D:\TrustedData\git\six-library-master\target
Platform                                 : win32
Checking for 'msvc' (c compiler)         : C:\TrustedApps\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\CL.exe
Checking for 'msvc' (c++ compiler)       : C:\TrustedApps\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\CL.exe
Checking for 64-bit system               : yes
System size                              : 64-bit
Checking for program matlab              : not found
Checking for program python              : C:\Python27\python.exe
Checking for python version              : 2.7.8

Successful. 

The I run 
\git\six-library-master>python waf
Got a compiler error when compiling the downsampler.c  
d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\downsampler.c(1029) : fatal error C1001: An internal error has occurred in the compiler.
(compiler file 'd:\agent\_work\3\s\src\vctools\compiler\utc\src\p2\main.c', line 187)
 To work around this problem, try simplifying or changing the program near the locations listed above.
Please choose the Technical Support command on the Visual C++
 Help menu, or open the Technical Support help file for more information
  CL!InvokeCompilerPassW()+0x92ae3
  CL!InvokeCompilerPassW()+0x91918

d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\downsampler.c(1029) : fatal error C1001: An internal error has occurred in the compiler.
(compiler file 'd:\agent\_work\3\s\src\vctools\compiler\utc\src\common\error.c', line 835)
 To work around this problem, try simplifying or changing the program near the locations listed above.
Please choose the Technical Support command on the Visual C++
 Help menu, or open the Technical Support help file for more information
SegmentWriter.c
SubWindow.c
d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\subwindow.c(53): warning C4100: 'error': unreferenced formal parameter
StreamIOWriteHandler.c
TRE.c
d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\tre.c(48): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
TRECursor.c
d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\trecursor.c(152): warning C4100: 'error': unreferenced formal parameter
TREPrivateData.c
d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\treprivatedata.c(150): warning C4100: 'error': unreferenced formal parameter
d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\treprivatedata.c(149): warning C4100: 'userData': unreferenced formal parameter
d:\trusteddata\git\six-library-master\externals\nitro\modules\c\nitf\source\treprivatedata.c(147): warning C4100: 'ht': unreferenced formal parameter
ArgumentParser.cpp
Waf: Leaving directory `D:\TrustedData\git\six-library-master\target'

I am not sure how to the correct the downsampler.c. Have you ever try 
MSVC 2017\Enterprise\VC\Tools\MSVC\14.16.27023? Is this a know problem?

Thanks!

JonathanMeans, 2019-03-18:
@GitGeoCoder 

It looks like I did not understand the MSVC version information correctly when I fixed this before. I'll put in a fix shortly. In the mean time, you can go to line 937 of Downsampler.c and remove the upper bound for the check there.

asylvest, 2019-03-18:
Thanks @JonathanMeans.  Can you also update the SIX manual when you get a chance..?

asylvest, 2019-03-18:
FYI @GitGeoCoder this has all been fixed in master by @JonathanMeans.  Let us know if you still have any trouble.

GitGeoCoder, 2019-03-18:
@JonathanMeans @asylvest Thanks again for the quick response. I have built and tested it successfully.

GitGeoCoder, 2019-03-19:
I get the croppedNitfs nitf files loaded right. But when loading some of my files  with "SICD:0.3.1", I got a validation error.  Any idea?

asylvest, 2019-03-19:
Whoa - that is an old SICD!  The earliest version of SICD that SIX supports is 0.4.0.  Any version older than that is from a draft version of the SICD spec... is it possible to update the software producing the SICD?

GitGeoCoder, 2019-03-19:
Tried 1.1.0, works!
In order to make the SICD/SIDD CSM plugin work, I have to copy the  schema into the upper directory of the plugin dll
,,/data/schema/six/*.xsd

It could be better if it's ok to put them into a sub directory of the plugin,  like data/schema/six/*.xsd. 
Just a suggestion.

asylvest, 2019-03-19:
OK cool.  Yeah the location the CSM plugin looks for the schemas is dictated by convention (MSP convention if I recall correctly), but glad you figured out where to put them.

closed 2020-09-21 by JDanielSmith