ngageoint/six-library #192: Example product in SIX compatible formats?

piyushrpt, 2017-06-15:
Hi,
    Is there an archive of example products generate using SIX/SICD/SIDD? I'm interested in finding workingexamples that converts data from current missions like TerraSAR-X, COSMO-SkyMed, Sentinel-1 or Radarsat-2 into one of the SIX compatible formats. In particular, I'm interested in using SLC products for interferometric applications and in evaluating the completeness of metadata fields for such applications. 

Piyush

asylvest, 2017-06-16:
There is a sample SICD from a public TerraSAR-X collect at https://github.com/ngageoint/six-library/wiki/Sample-SICDs.  A lot of the issues posting more datasets are related to what's publicly available and/or ITAR concerns.

khavernathy, 2020-01-22:
Are there SIX-compatible CPHD or CPHD03 files anywhere?

When I use some of the `.../cphd03/tests` executables on files I have, I get `Unknown exception`.

JonathanMeans, 2020-01-23:
I'm looking into more samples that could be made available.

In the mean time, if you can share what you have I might be able to dig into the error.

closed 2020-11-03 by JDanielSmith