ngageoint/six-library #216: Shared library compilation fails

apeternier, 2017-12-13:
I'm trying to build the code with the --shared option set to Waf, but compilation fails with a problem related to the libxerces.def file. Static building works well, on the other hand. 
This problem is the same both under Windows and Linux. 
Since the library is released under LGPL v3.0, we can only use this software as a dynamic library within a commercial, closed-source project. 
Otherwise, the inability to generate a .dll/.so distribution of six-library quite contradicts the statement "[six-library] allows applications that are not open source to make use of the library without penalty".

asylvest, 2017-12-13:
Thanks for letting us know.  We are actually just starting to look at updating the build system to support this for NITRO due to a similar request; once this is finished, we'll ripple it up to SIX as well.

JonathanMeans, 2017-12-13:
@apeternier I'm surprised that this is failing on Linux. Can you try removing your install directory and doing

```
waf clean configure --prefix=install install
waf configure --prefix=install --shared install
```
Also see [here](https://github.com/mdaus/nitro/issues/62#issuecomment-351385630) for a hack that may get you going on Windows until we have a real fix.

apeternier, 2017-12-13:
Thank you for the quick replies. Just to expand a little: I only need the sicd and sidd libraries for reading/writing such files. 

What is triggering the problem is as follows (starting from a fresh git clone):

1) python waf configure --prefix=buildOut --enable-64bit --shared
2) python waf install --target=six.sicd-c++

And the output is:
Waf: Entering directory `F:\six-library\target'
Could not load module numpy. Disabling Python installation.
Waf: Leaving directory `F:\six-library\target'
invalid def file 'lib/libxerces.def'

I tried the workaround you linked, but it does not generate the .dll I'm looking for (I expect that a --shared build of six.sicd-c++ and six.sidd-c++ should build something with a similar name and .dll at the end). 

A clean install under Linux (as you asked) generated a lib directory full of missing symbolic links to .so files that were not there.

closed 2020-11-03 by JDanielSmith