ngageoint/six-library #278: SIXSensorModel::getValidImageRange() and SIXSensorModel::getValidHeightRange() methods do not have valid implementations.

johnpdpkarp, 2019-05-28:
SIXSensorModel::getValidHeightRange() should return the minimum and maximum height for which the model is valid.  For some sensor models, these values does have much impact.  For active sensors, the MINIMUM height is especially important, as it is entirely possible that the range arc does not intersect the Earth for heights below a certain value.

Similarly, csm::RasterGM is always supposed to provide results relative to "full image" coordinates.  If a particular SICD / SIDD comprises a sub-image, and the metadata can only accomodate that sub-image, that should be reflected in getValidImageRange().  Otherwise, getValidImageRange() should reflect  the size of the full image.   Currently both of these methods are coded to return values from (-99999, 99999) .

asylvest, 2019-05-28:
I believe the guidance we received originally was that these were low priority methods to implement and the 99999 value was the recommended way to essentially say "everything".

We can provide your comments to the customer to recommend that we provide legit implementations for these.

johnpdpkarp, 2019-05-28:
That is unfortunate.  getValidHeightRange() was specifically added to the API to support active sensors like SAR which have a very definite valid "minimum" height.  The "guidance" value is probably okay for the maximum height...
getValidImageRange() should never be returning these default values.  If the whole image is valid, it should be equivalent to asking for getImageStart() and getImageSize().  In some instances, the sensor model may be valid over the entire image, or just over some AOI.  For example, if you had a scan image that was 2000x2000 pixels which was chipped out of a larger image (3000x10000) starting at location (1000,1000), and the metadata provided only provided information to process over that chip, getImageStart() should give (1000,1000), getImageSize() should give (2000,2000), and getValidImageRange() should give  (1000,1000),(3000,3000).   If the metadata actually supported the entire image, getImageStart() and getImageSize() would be the same, but getValidImageRange() could actually give (0,0), (3000,10000).  The purpose of the method is to let the user know (without having to depend on a warning or error being returned), which image coordinates can be successfully converted to ground.

JonathanMeans, 2019-09-10:
Addressed in https://github.com/ngageoint/six-library/pull/285

closed 2020-11-03 by JDanielSmith