ngageoint/six-library #279: Chipped Images

johnpdpkarp, 2019-05-28:
The CSM API is intended to work with full image coordinates.  It appears that, as currently implemented, if presented with an image chip, the SICD sensor model offsets the input pixel coordinate based on the FirstRow and FirstCol as defined in the XML.  This is not correct.  These would be appropriate values to return via the overloaded getImageStart() method.   Tools exploiting CSM are expected to handle the conversion from chipped coordinates to full image coordinates.

asylvest, 2019-05-29:
@johnpdpkarp you are correct that the current implementation for SICD and SIDD chipped products is that all coordinates are treated as with respect to the chipped product, not the original full image product.  So are you saying that, for example, `groundToImage()` should take in the ECEF location and return the pixel location in full image coordinates (and `getImageStart()` returns nonzero to reflect the chipping)?

johnpdpkarp, 2019-05-29:
The intent of the CSM API is for the sensor model to operate in "full image" coordinates.  Whatever software tools exploit CSM are supposed to address how full image coordinates relate to the image chip, and imageToGround() should be passing in "full image" coordinates and groundToImage() should be returning "full image" coordinates.  Since the SICD and SIDD sensor models use the entire SICD/SIDD XML structure as their "state", and thus know about the FirstRow and FirstCol values, it would make sense for getImageStart() to return these values.  This method is really just intended to be "informational" and should have no bearing on the results of imageToGround() or groundToImage().

closed 2020-11-03 by JDanielSmith