ngageoint/six-library #295: static NITFReadControl::getDataType always returns NOT_SET unless a NITFReadControl is instantiated

creffett, 2019-12-09:
Apologies for the vagueness of this report, the computer where I found this issue doesn't have GitHub access. I'm trying to call the static method six::NITFReadControl::getDataType on a Nitro nitf::Record object. When I use that method on a well-formed SICD or SIDD, it always returns six::DataType::NOT_SET unless I create a NITFReadControl first, in which case it returns the correct DataType. Note that I don't need to call the static method through the object, I just need to create one. I am using the latest master branch.

JonathanMeans, 2019-12-09:
Looks like there's a lot of implicit global static state flying around here. Until I work out a proper solution, you can work around this by calling `six::loadXmlDataContentHandler()` (declared in `six/Utilities.h`) before calling `getDataType()`

JonathanMeans, 2019-12-10:
Could you email me an example of what you're doing? Constructing a `NITFReadControl` vs calling `getDataType` statically isn't making a difference in any of my tests.

closed 2020-01-21 by JonathanMeans