ngageoint/six-library #300: how do i use the c++ samples?

khavernathy, 2020-01-17:
Not able to compile any of the sample c++ code after installing six-library.

`cd six-library/six/modules/c++/samples`
`g++ -std=c++11 extract_cphd_xml.cpp`

I get `undefined reference` to various classes no matter which sample i try to compile, e.g. below.

Grateful for any help.

```
/tmp/ccEra1O8.o: In function `main':
extract_cphd_xml.cpp:(.text+0xc5): undefined reference to `cli::ArgumentParser::ArgumentParser()'
extract_cphd_xml.cpp:(.text+0x108): undefined reference to `cli::ArgumentParser::setDescription(std::string const&)'
extract_cphd_xml.cpp:(.text+0x225): undefined reference to `cli::ArgumentParser::addArgument(std::string const&, std::string const&, cli::Action, std::string const&, std::string const&, int, int, bool)'
extract_cphd_xml.cpp:(.text+0x3c7): undefined reference to `cli::ArgumentParser::addArgument(std::string const&, std::string const&, cli::Action, std::string const&, std::string const&, int, int, bool)'
extract_cphd_xml.cpp:(.text+0x569): undefined reference to `cli::ArgumentParser::addArgument(std::string const&, std::string const&, cli::Action, std::string const&, std::string const&, int, int, bool)'
extract_cphd_xml.cpp:(.text+0x70b): undefined reference to `cli::ArgumentParser::addArgument(std::string const&, std::string const&, cli::Action, std::string const&, std::string const&, int, int, bool)'
extract_cphd_xml.cpp:(.text+0x891): undefined reference to `cli::ArgumentParser::addArgument(std::string const&, std::string const&, cli::Action, std::string const&, std::string const&, int, int, bool)'
extract_cphd_xml.cpp:(.text+0x952): undefined reference to `cli::ArgumentParser::parse(int, char const**)'
extract_cphd_xml.cpp:(.text+0xda5): undefined reference to `sys::Path::basename(std::string const&, bool)'
extract_cphd_xml.cpp:(.text+0xe43): undefined reference to `sys::OSUnix::getNumCPUs() const'
extract_cphd_xml.cpp:(.text+0xe67): undefined reference to `cphd::CPHDReader::CPHDReader(std::string const&, unsigned long, std::vector<std::string, std::allocator<std::string> > const&, std::shared_ptr<logging::Logger>)'
extract_cphd_xml.cpp:(.text+0xea4): undefined reference to `cphd::CPHDXMLControl::CPHDXMLControl(logging::Logger*, bool)'
extract_cphd_xml.cpp:(.text+0xeda): undefined reference to `cphd::CPHDXMLControl::toXMLString(cphd::Metadata const&, std::vector<std::string, std::allocator<std::string> > const&, bool)'
extract_cphd_xml.cpp:(.text+0xf38): undefined reference to `io::FileOutputStreamOS::FileOutputStreamOS(std::string const&, int)'
extract_cphd_xml.cpp:(.text+0xfbb): undefined reference to `cphd::CPHDXMLControl::~CPHDXMLControl()'
extract_cphd_xml.cpp:(.text+0x1033): undefined reference to `cli::ArgumentParser::~ArgumentParser()'
extract_cphd_xml.cpp:(.text+0x165d): undefined reference to `cphd::CPHDXMLControl::~CPHDXMLControl()'
extract_cphd_xml.cpp:(.text+0x16fd): undefined reference to `cli::ArgumentParser::~ArgumentParser()'
/tmp/ccEra1O8.o: In function `__static_initialization_and_destruction_0(int, int)':
extract_cphd_xml.cpp:(.text+0x18a0): undefined reference to `sys::MutexPosix::MutexPosix()'
extract_cphd_xml.cpp:(.text+0x18af): undefined reference to `sys::MutexPosix::~MutexPosix()'
/tmp/ccEra1O8.o: In function `except::Throwable::toString() const':
extract_cphd_xml.cpp:(.text._ZNK6except9Throwable8toStringEv[_ZNK6except9Throwable8toStringEv]+0xfa): undefined reference to `except::operator<<(std::ostream&, except::Trace const&)'
/tmp/ccEra1O8.o: In function `except::Exception::Exception(except::Context const&)':
extract_cphd_xml.cpp:(.text._ZN6except9ExceptionC2ERKNS_7ContextE[_ZN6except9ExceptionC5ERKNS_7ContextE]+0x33): undefined reference to `except::Throwable::Throwable(except::Context)'
/tmp/ccEra1O8.o: In function `sys::TimeStamp::local() const':
extract_cphd_xml.cpp:(.text._ZNK3sys9TimeStamp5localEv[_ZNK3sys9TimeStamp5localEv]+0x19): undefined reference to `sys::LocalDateTime::LocalDateTime()'
extract_cphd_xml.cpp:(.text._ZNK3sys9TimeStamp5localEv[_ZNK3sys9TimeStamp5localEv]+0x5e): undefined reference to `sys::DateTime::format(std::string const&) const'
/tmp/ccEra1O8.o: In function `sys::OSUnix::OSUnix()':
extract_cphd_xml.cpp:(.text._ZN3sys6OSUnixC2Ev[_ZN3sys6OSUnixC5Ev]+0x14): undefined reference to `sys::AbstractOS::AbstractOS()'
extract_cphd_xml.cpp:(.text._ZN3sys6OSUnixC2Ev[_ZN3sys6OSUnixC5Ev]+0x1f): undefined reference to `vtable for sys::OSUnix'
/tmp/ccEra1O8.o: In function `sys::OSUnix::~OSUnix()':
extract_cphd_xml.cpp:(.text._ZN3sys6OSUnixD2Ev[_ZN3sys6OSUnixD5Ev]+0x13): undefined reference to `vtable for sys::OSUnix'
extract_cphd_xml.cpp:(.text._ZN3sys6OSUnixD2Ev[_ZN3sys6OSUnixD5Ev]+0x1f): undefined reference to `sys::AbstractOS::~AbstractOS()'
/tmp/ccEra1O8.o: In function `io::StandardOutStream::StandardOutStream()':
extract_cphd_xml.cpp:(.text._ZN2io17StandardOutStreamC2Ev[_ZN2io17StandardOutStreamC5Ev]+0x1f): undefined reference to `vtable for io::StandardOutStream'
/tmp/ccEra1O8.o: In function `cphd::Data::~Data()':
extract_cphd_xml.cpp:(.text._ZN4cphd4DataD2Ev[_ZN4cphd4DataD5Ev]+0x13): undefined reference to `vtable for cphd::Data'
/tmp/ccEra1O8.o: In function `cphd::FileHeader::~FileHeader()':
extract_cphd_xml.cpp:(.text._ZN4cphd10FileHeaderD2Ev[_ZN4cphd10FileHeaderD5Ev]+0x13): undefined reference to `vtable for cphd::FileHeader'
/tmp/ccEra1O8.o: In function `cli::Argument* cli::Argument::setDefault<bool>(bool)':
extract_cphd_xml.cpp:(.text._ZN3cli8Argument10setDefaultIbEEPS0_T_[_ZN3cli8Argument10setDefaultIbEEPS0_T_]+0x3f): undefined reference to `cli::Argument::setDefault(cli::Value*, bool)'
/tmp/ccEra1O8.o: In function `cli::Argument* cli::Argument::setDefault<char const*>(char const*)':
extract_cphd_xml.cpp:(.text._ZN3cli8Argument10setDefaultIPKcEEPS0_T_[_ZN3cli8Argument10setDefaultIPKcEEPS0_T_]+0x3f): undefined reference to `cli::Argument::setDefault(cli::Value*, bool)'
/tmp/ccEra1O8.o: In function `bool cli::Value::at<bool>(unsigned long) const':
extract_cphd_xml.cpp:(.text._ZNK3cli5Value2atIbEET_m[_ZNK3cli5Value2atIbEET_m]+0x60): undefined reference to `str::format(char const*, ...)'
extract_cphd_xml.cpp:(.text._ZNK3cli5Value2atIbEET_m[_ZNK3cli5Value2atIbEET_m]+0x192): undefined reference to `bool str::toType<bool>(std::string const&)'
/tmp/ccEra1O8.o: In function `std::string cli::Value::at<std::string>(unsigned long) const':
extract_cphd_xml.cpp:(.text._ZNK3cli5Value2atISsEET_m[_ZNK3cli5Value2atISsEET_m]+0x6a): undefined reference to `str::format(char const*, ...)'
extract_cphd_xml.cpp:(.text._ZNK3cli5Value2atISsEET_m[_ZNK3cli5Value2atISsEET_m]+0x1a9): undefined reference to `std::string str::toType<std::string>(std::string const&)'
/tmp/ccEra1O8.o: In function `sys::LocalDateTime::~LocalDateTime()':
extract_cphd_xml.cpp:(.text._ZN3sys13LocalDateTimeD2Ev[_ZN3sys13LocalDateTimeD5Ev]+0x13): undefined reference to `vtable for sys::LocalDateTime'
extract_cphd_xml.cpp:(.text._ZN3sys13LocalDateTimeD2Ev[_ZN3sys13LocalDateTimeD5Ev]+0x1f): undefined reference to `sys::DateTime::~DateTime()'
collect2: error: ld returned 1 exit status
```

JonathanMeans, 2020-01-17:
How are you installing it?

If you have Python, the easiest way would be to use the waf build system, which comes with the code. Try

`python waf configure --prefix=install install --target=extract_cphd_xml`

See [this README](https://github.com/ngageoint/six-library/tree/master/externals/coda-oss) for documentation on using waf.

khavernathy, 2020-01-17:
this helps, ty. Example c++ files compile and run fine now.

I was using 
`sudo ./waf configure; sudo ./waf install`

from the repo's root directory

closed 2020-01-17 by JonathanMeans