ngageoint/six-library #302: What is the schema path?

khavernathy, 2020-01-17:
I'm trying to use an example from SIX `test_round_trip`:
```
dmf@blfldmf:~/six-library/target/six/modules/c++/cphd/tests$ ./test_round_trip  -t 6 /scratch/cphd_mess/tester.cphd  out.cphd    
ParseException: usage: ./test_round_trip [-t NUM] CPHD CPHD XSD    
missing required argument: [schema] 
```

I need a "schema". What/where is that?

khavernathy, 2020-01-17:
Reading the manual a bit and `grep`ing for `schema` led me to find

`target/six/conf/setup.py` after installing..

I did `sudo python setup.py install` and `grep`ed around again to find
`target/six/conf/build/lib/pysix/schemas/SICD_schema_V1.2.0_2016_04_12.xsd`

I used that as the [schema] argument and the error vanished..

JonathanMeans, 2020-01-18:
The schemas should get installed to `<install>/conf/schema/six/`. If you're not seeing it, try a `waf install` with no arguments.

You'll want to pass the directory containing the schemas, and it will then automatically choose the correct one for schema validation.

closed 2020-02-21 by JonathanMeans