ngageoint/six-library #305: A basic implementation of using/linking six c++ library?

khavernathy, 2020-01-22:
I want to implement the six library in existing software to read/write cphd files

I've successfully installed using `./waf install` already into the default `/usr/local` destination

I've copied six-library/six/modules/c++/cphd03/tests/print_cphd_header.cpp to a localdir `~/localdir`.

When I try to compile it alone I get the following:
```
me@machine:~/localdir$ g++ print_cphd_header.cpp -std=c++11 -o go                                     
/tmp/ccN3Jfoz.o: In function `main':
print_cphd_header.cpp:(.text+0x17a): undefined reference to `sys::Path::basename(std::string const&, bool)'
print_cphd_header.cpp:(.text+0x2b1): undefined reference to `logging::setupLogger(std::string const&, std::string const&, std::string const&, std::string const&, unsigned long, unsigned long)'
print_cphd_header.cpp:(.text+0x353): undefined reference to `cphd03::CPHDReader::CPHDReader(std::string const&, unsigned long, std::shared_ptr<logging::Logger>)'
print_cphd_header.cpp:(.text+0x39f): undefined reference to `cphd03::operator<<(std::ostream&, cphd03::FileHeader const&)'
print_cphd_header.cpp:(.text+0x3b7): undefined reference to `cphd03::operator<<(std::ostream&, cphd03::Metadata const&)'
print_cphd_header.cpp:(.text+0x3cf): undefined reference to `cphd03::operator<<(std::ostream&, cphd03::VBM const&)'
/tmp/ccN3Jfoz.o: In function `__static_initialization_and_destruction_0(int, int)':
print_cphd_header.cpp:(.text+0x6ea): undefined reference to `sys::MutexPosix::MutexPosix()'
print_cphd_header.cpp:(.text+0x6f9): undefined reference to `sys::MutexPosix::~MutexPosix()'
/tmp/ccN3Jfoz.o: In function `cphd03::FileHeader::~FileHeader()':
print_cphd_header.cpp:(.text._ZN6cphd0310FileHeaderD2Ev[_ZN6cphd0310FileHeaderD5Ev]+0x13): undefined reference to `vtable for cphd03::FileHeader'
collect2: error: ld returned 1 exit status
```

How do I explicitly link libraries etc. to get the file to compile on its own?

JonathanMeans, 2020-01-22:
`waf dumplib --target=cphd03-c++` (adjust the target as appropriate) will give you the link flags you need to pass to gcc in the correct order.

You can also combine targets. E.g.

    >> waf dumplib --target=six-c++
    -lsix-c++.2.2.9 -lscene-c++.2.2.9 -lnitf-c++.2.8 -lxml.lite-c++.1.2 [...]

    >> waf dumplib --target=six-c++,cphd03-c++
    -lcphd03-c++ -lcphd-c++.2.2.9 -lsix.sicd-c++.2.2.9 -lsix-c++.2.2.9 -lscene-c++.2.2.9 [...]

khavernathy, 2020-01-22:
Thanks, this helps

But when I try

`g++  -L/usr/local/lib -I/usr/local/include -lcphd03-c++ -lcphd-c++.2.2.9 -lsix.sicd-c++.2.2.9 -lsix-c++.2.2.9 -lscene-c++.2.2.9 -lnitf-c++.2.8 -lxml.lite-c++.1.2 -llogging-c++.1.1 -lio-c++.1.0 -lmath.linear-c++.0.2 -lpolygon-c++.1.0 -lmt-c++.1.1 -lmem-c++.1.0 -lmath-c++.0.1 -lsys-c++.1.2 -lstr-c++.1.0 -lexcept-c++.1.0 -lXML_DATA_CONTENT-static-c.2.8 -lnitf-c.2.8 -lnrt-c -lxerces-c -ldl -lrt -lm print_cphd_header.cpp  -std=c++11 -o ./go`

I still get the error above. I've confirmed the files are in `/usr/local/include` ..

JonathanMeans, 2020-01-22:
You need to have the libraries after the source file.

`g++ -std=c++11 print_cphd_header.cpp -L/usr/local/lib -I/usr/local/include -lcphd03-c++ -lcphd-c++.2.2.9 -lsix.sicd-c++.2.2.9 -lsix-c++.2.2.9 -lscene-c++.2.2.9 -lnitf-c++.2.8 -lxml.lite-c++.1.2 -llogging-c++.1.1 -lio-c++.1.0 -lmath.linear-c++.0.2 -lpolygon-c++.1.0 -lmt-c++.1.1 -lmem-c++.1.0 -lmath-c++.0.1 -lsys-c++.1.2 -lstr-c++.1.0 -lexcept-c++.1.0 -lXML_DATA_CONTENT-static-c.2.8 -lnitf-c.2.8 -lnrt-c -lxerces-c -ldl -lrt -lm -o ./go`

khavernathy, 2020-01-22:
Awesome, thanks. In my case the library `-lpthread` was needed too.

So the working compilation looked like this:

`g++ -std=c++11 print_cphd_header.cpp -L/usr/local/lib -I/usr/local/include -lcphd03-c++ -lcphd-c++.2.2.9 -lsix.sicd-c++.2.2.9 -lsix-c++.2.2.9 -lscene-c++.2.2.9 -lnitf-c++.2.8 -lxml.lite-c++.1.2 -llogging-c++.1.1 -lio-c++.1.0 -lmath.linear-c++.0.2 -lpolygon-c++.1.0 -lmt-c++.1.1 -lmem-c++.1.0 -lmath-c++.0.1 -lsys-c++.1.2 -lstr-c++.1.0 -lexcept-c++.1.0 -lXML_DATA_CONTENT-static-c.2.8 -lnitf-c.2.8 -lnrt-c -lxerces-c -ldl -lrt -lm -lpthread -o ./go`

closed 2020-01-23 by JonathanMeans