ngageoint/six-library #328: Fails to compile on 10.15.3 Catalina

CobaltGray, 2020-03-11:
When trying to compile on OSX 10.15.3 the compilation throughs the error:

`Build failed
 -> task in 'mt-c++' failed (exit status 1): 
	{task 140367451013520: cxx ThreadGroup.cpp -> ThreadGroup.cpp.1.o}
['/usr/local/bin/g++', '-fPIC', '-std=c++11', '-Wall', '-Wno-deprecated-declarations', '-O3', '-m64', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/mt/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/mt/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/include', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/math/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/math/include', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/mem/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/mem/include', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/sys/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/sys/include', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/str/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/str/include', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/except/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/except/include', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/config/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/config/include', '-I/Users/darren/Development/six-library/target/externals/coda-oss/modules/c++/types/include', '-I/Users/darren/Development/six-library/externals/coda-oss/modules/c++/types/include', '-D_FILE_OFFSET_BITS=64', '-D_LARGEFILE_SOURCE', '-DSWIG_PYTHON_SILENT_MEMLEAK', '-DPYTHONDIR="/usr/local/lib/python2.7/site-packages"', '-DPYTHONARCHDIR="/usr/local/lib/python2.7/site-packages"', '-DHAVE_PYTHON_H=1', '-DNITF_DEFAULT_PLUGIN_PATH="/usr/local/share/nitf/plugins"', '-DDEFAULT_SCHEMA_PATH="/usr/local/conf/schema/six"', '-D_LARGEFILE_SOURCE=1', '-DSWIG_PYTHON_SILENT_MEMLEAK=1', '-DHAVE_PYTHON_H', '../externals/coda-oss/modules/c++/mt/source/ThreadGroup.cpp', '-c', '-o', 'externals/coda-oss/modules/c++/mt/source/ThreadGroup.cpp.1.o']`

It looks like CPUAffinityInitializerLinux is not being defined. In ...mt/CPUAdffinityInitialiser.h : 
`#include <mt/AbstractCPUAffinityInitializer.h>

#if defined(WIN32)
#include <mt/CPUAffinityInitializerWin32.h>
namespace mt
{
typedef CPUAffinityInitializerWin32 CPUAffinityInitializer;
}
#endif

#if !defined(__APPLE_CC__)
#if defined(__linux) || defined(__linux__)
#include <mt/CPUAffinityInitializerLinux.h>
namespace mt
{
typedef CPUAffinityInitializerLinux CPUAffinityInitializer;
}
#endif
#endif
`
so no define if using __APPLE_CC__

JonathanMeans, 2020-03-11:
Have you confirmed that this change works for you?

Supporting Mac is not a priority for us (and I don't have a way to test this in any case), but if you have a fix I can probably get it worked in.

CobaltGray, 2020-03-14:
No sorry couldn't get it working yet. There are more subtleties than just hacking #define loops around the lack of #ifdef __APPLE_CC__ that defeated me.

closed 2020-11-24 by JDanielSmith