ngageoint/six-library #333: SIX 3.0 C++11 requirement

khanh799, 2020-04-09:
We have a really old Linux server running CentOS 6 and the latest compiler we can run on it is 4.4.7 which doesn't support C++11 standard.  Is it possible to get a version of SIX 3.0 that doesn't need this requirement?

JonathanMeans, 2020-04-16:
We hadn't planned to support this. How critical is the need?

If you want to make the changes in your own fork, here are a couple of the tricky places:

- externals/coda-oss/build/build.py: line 908 add the `c++11` flag. Line 1304 causes the build to fail without C++11
- There's a `mem::SharedPtr` that inherits from `std::shared_ptr`. Historically, this was for compatability with a custom non-c++11 `SharedPtr`. The code's been deleted, but you can find the most recent checkout [here](https://github.com/mdaus/coda-oss/blob/d41ee9b79893e108734c925acd449ebfad990a94/modules/c%2B%2B/mem/include/mem/SharedPtrLegacy.h)
- [This PR](https://github.com/mdaus/coda-oss/pull/328/files) should have all the "hard" stuff.

khanh799, 2020-04-18:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi Jonathan,

This is now OBE.  I’m able to get a new workstation to compile this.

Trying it out, I see the library is not able to read Complex Int 16.  Are you aware of this?

Khanh Ho
Engineer, Enterprise Engineering
Khanh.Ho@centauricorp.com<mailto:Khanh.Ho@centauricorp.com>
15020 Conference Center Drive
Chantilly, VA  20151
Phone:   703-378-1282 (Centauri)
Phone:   571-557-2743 (customer site)
Mobile:  703-623-4930
www.centauricorp.com<http://www.centauricorp.com/>

JonathanMeans, 2020-04-20:
I was just able to read the 16-bit complex int SICD on the [wiki](https://github.com/ngageoint/six-library/wiki/Sample-SICDs): https://six-library.s3.amazonaws.com/sicd_example_RMA_RGZERO_RE16I_IM16I.nitf

Do you have an example of a SICD that's failing? Or do you mean that the pixels are wrong when you read it?

khanh799, 2020-04-20:
Do you have an email that I can use to send FTP information to?

Khanh Ho
Engineer, Enterprise Engineering
Khanh.Ho@centauricorp.com<mailto:Khanh.Ho@centauricorp.com>
15020 Conference Center Drive
Chantilly, VA  20151
Phone:   703-378-1282 (Centauri)
Phone:   571-557-2743 (customer site)
Mobile:  703-623-4930
www.centauricorp.com<http://www.centauricorp.com/>

JonathanMeans, 2020-04-20:
You can send it to jonathan.means@maxar.com

closed 2020-09-21 by JDanielSmith