ngageoint/six-library #534: Slow performance reading and writing AMP8I_PHS8I

andrew-hardin, 2022-01-19:
I have a simple processing chain that looks like this:
```
1. Read AMP8I_PHS8I as std::complex<float>.
2. Modify std::complex<float>.
3. Write std::complex<float> as AMP8I_PHS8I.
```
The bottleneck on this process is reading and writing the pixel content, specifically the methods tied to the conversions between AMP8I_PHS8I and `std::complex<float>`. Is there any interest in future pull requests that improve the performance of those methods?

JDanielSmith, 2022-01-19:
I've set things up to do all the AMP8I_PHS8I -> `std::complex<float>` conversion at once (instead of pixel-by-pixel) for a particular data-set (at least that's the intention).  From what I can tell, the real bottleneck is the *KDTree* structure I'm using to find a (amp, phs) pair for any given `std::complex<float>`. If you've got a different approach, that would be great! :-)

andrew-hardin, 2022-01-19:
Your note about the KDTree matches some functions identified by `perf`. There may not be a better alternative, but I can't say for sure because I haven't fully wrapped my head around what's going on yet.

Are there any unmerged changes on a branch that I should build on, or, is everything related already merged with `master`?

JDanielSmith, 2022-01-20:
The only thing not in **master** is your most recent PR; although I would work in the **feature/8AMPI_PHSI** branch.

andrew-hardin, 2022-01-20:
Thank you for your guidance on where to start - I'll begin taking a look at optimizations to improve performance.

JDanielSmith, 2022-01-20:
The first thing to try is enabling the multi-threaded implementation by passing a positive value to `ImageData::to_AMP8I_PHS8I()`; see https://github.com/ngageoint/six-library/blob/master/six/modules/c%2B%2B/six.sicd/source/ImageData.cpp

JDanielSmith, 2022-01-20:
I've exposed the `cutoff` parameter which controls multi-threading for `to_AMP8I_PHS8I()` through the various layers of code; I've also turned on multi-threading by default.

See the latest changes in the https://github.com/ngageoint/six-library/tree/feature/8AMPI_PHSI branch.

JDanielSmith, 2022-01-26:
This should be better with multi-threading enabled by default: https://github.com/ngageoint/six-library/pull/536

closed 2022-01-26 by JDanielSmith